gopro/OpenGoPro #217: Bluetooth connection to paired device

kobezf, 2022-08-05:
I have the embedded linux device with bluetooth sensor. I connected to a GoPRO camera as a new client and paired it. After the disconnection, I am not able to connect to it without pairing mode. I always have to open GoPro Quik App menu to connect to the GOPRO. This can be reproduced only on the embedded device, as from the laptop I can successfully connect to already paired device without any manipulations with the camera. I compared the bluetooth dump and cannot see any differences in connection. Here is the dump from my embedded device:
Successful connection till get used feature:
< 01 0D 20 19 60 00 60 00 00 01 6A B0 F7 9D EA D8 00 18 00 28 
  00 00 00 2A 00 00 00 00 00 
> 04 0F 04 00 01 0D 20 
> 04 3E 13 01 00 01 04 00 01 6A B0 F7 9D EA D8 28 00 00 00 2A 
  00 01 
< 01 16 20 02 01 04 
> 04 0F 04 00 01 16 20 
> 04 3E 0C 04 **00** 01 04 FF 00 00 00 00 00 00 00

Failed connection:
< 01 0D 20 19 60 00 60 00 00 01 6A B0 F7 9D EA D8 00 18 00 28 
  00 00 00 2A 00 00 00 00 00 
> 04 0F 04 00 01 0D 20 
> 04 3E 13 01 00 01 04 00 01 6A B0 F7 9D EA D8 28 00 00 00 2A 
  00 01 
< 01 16 20 02 01 04 
> 04 0F 04 00 01 16 20 
> 04 3E 0C 04 **3E** 01 04 00 00 00 00 00 00 00 00

there are no any differences in the connections, but I don't understand why I cannot connect to GoPRO without opening GoPro Quik App menu. Do you have any ideas? Thank you

tcamise-gpsw, 2022-08-08:
Hello. I have no idea what these bytes mean. It would be helpful if you can translate them into a human-readable log.

Based on your description, it sounds like the embedded device is doing something strange. Perhaps it is not actually disconnecting?

One thing you can try is to use either [LightBlue ](https://apps.apple.com/us/app/lightblue/id557428110)(for iOS) or [nrfConnect ](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_US&gl=US)(For Android) to compare the advertising data after the disconnections. The advertisements should look exactly same in both the laptop and embedded Linux cases.

kobezf, 2022-08-09:
I tried rebooting the camera and still couldn't connect to it. I deleted all connections to the GOPRO, connected to it, paired again and disconnected. After that, I tried to connect again without putting the camera into discovery mode, but the connection failed. Here is the output of btmon:
```< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
        Type: Passive (0x00)
        Interval: 60.000 msec (0x0060)
        Window: 30.000 msec (0x0030)
        Own address type: Public (0x00)
        Filter policy: Ignore not in white list (0x01)
> HCI Event: Command Complete (0x0e) plen 4
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 35
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: D8:EA:9D:F7:B0:6A (Static)
        Data length: 23
        Flags: 0x02
          LE General Discoverable Mode
        16-bit Service UUIDs (partial): 1 entry
          GoPro, Inc. (0xfea6)
        Company: GoPro, Inc. (754)
          Data: 0201393f006ab0f79dead80f
        RSSI: -49 dBm (0xcf)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Create Connection (0x08|0x000d) plen 25
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: White list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: D8:EA:9D:F7:B0:6A (Static)
        Own address type: Public (0x00)
        Min connection interval: 30.00 msec (0x0018)
        Max connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 1025
        Role: Master (0x00)
        Peer address type: Random (0x01)
        Peer address: D8:EA:9D:F7:B0:6A (Static)
        Connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Master clock accuracy: 0x01
@ MGMT Event: Device Connected (0x000b) plen 36
        LE Address: D8:EA:9D:F7:B0:6A (Static)
        Flags: 0x00000000
        Data length: 23
        Flags: 0x02
          LE General Discoverable Mode
        16-bit Service UUIDs (partial): 1 entry
[... 280 lines cut ...]
        Count: 1
> ACL Data RX: Handle 3585 flags 0x02 dlen 27
> ACL Data RX: Handle 3585 flags 0x01 dlen 19
      ATT: Read By Group Type Response (0x11) len 41
        Attribute data length: 20
        Attribute group list: 2 entries
        Handle range: 0x0042-0x0047
        UUID: Vendor specific (b5f90090-aa8d-11e3-9046-0002a5d5c51b)
        Handle range: 0x0048-0x0053
        UUID: Vendor specific (b5f90080-aa8d-11e3-9046-0002a5d5c51b)
< ACL Data TX: Handle 3585 flags 0x00 dlen 11
      ATT: Read By Group Type Request (0x10) len 6
        Handle range: 0x0054-0xffff
        Attribute group type: Primary Service (0x2800)
> HCI Event: Number of Completed Packets (0x13) plen 5
        Num handles: 1
        Handle: 3585
        Count: 1
> ACL Data RX: Handle 3585 flags 0x02 dlen 9
      ATT: Error Response (0x01) len 4
        Read By Group Type Request (0x10)
        Handle: 0x0054
        Error: Attribute Not Found (0x0a)
```
I still don't know what is wrong.

ladyeklipse, 2022-08-10:
Same here, on Windows 10 desktop PC, while using an ASUS USB-BT400 dongle, I cannot connect to GoPro without setting GoPro manually into advertising mode (connect - quik app). Strangely enough, in "Bluetooth & Other Devices" section in Windows 10 settings, I can see my GoPro and it says "Paired". For some reason it says 0% battery, which is not true.
I can follow all the Python tutorials, but only when I set GoPro to pair manually every time before launching a script.

tcamise-gpsw, 2022-08-24:
For my own reference when coming back to this: the bug is that, in order to reconnect a previously paired GoPro, the GoPro must be manually set into pairing mode.

This really seems to me to be related to the Bluetooth hardware / driver.

The first step here is that you need to verify if the GoPro is advertising in the fail state when it doesn't connect.  Please see my steps [here ](https://github.com/gopro/OpenGoPro/issues/217#issuecomment-1208477209) to do so. 

In the meantime I can try to replicate. @ladyeklipse I am also on Windows 10 so I can try testing with the tutorials but I have gone through this recently with no problems. If I can replicate, I can get a sniffer capture which will tell us what is happening.

kobezf, 2022-08-25:
@tcamise-gpsw Do I understand correctly, that I should connect camera to the laptop, then disconnect and look in [LightBlue](https://apps.apple.com/us/app/lightblue/id557428110) to checking for advertising after disconnection? I tried to do this and it seems that the advertising is the same. The only difference is in KCBAdvDataRxPrimaryPHY(for laptop it is 129, for embedded device is 1). Here is the results:
From the embeded device:
[image]

From laptop
[image]

tcamise-gpsw, 2022-08-25:
Thanks for this. I was able to replicate today and have a potential fix: #223 

Are you able to test from the `improve_camera_discovery` branch?

kobezf, 2022-08-26:
@tcamise-gpsw  I installed the package from this branch and got result:
from laptop I was able to connect without making camera discoverable, but from the embedded device I got an error:
```
WARNING  09:08:55.125 Failed to cancel connection (/org/bluez/hci0/dev_D8_EA_9D_F7_B0_6A): 'NoneType' object has no attribute 'call'                                                           client.py:299
WARNING  09:08:55.168 Connection failed during establishment..                                                                                                                          bleak_wrapper.py:259
```
It looks like the cause was not the library. Now I think the reason is the bluetooth sensor on my device.

tcamise-gpsw, 2022-08-26:
I have an ubuntu PC I can test with. This will at least verify the bleak driver is working correctly on *Nix

tcamise-gpsw, 2022-08-26:
Everything looks good on my ubuntu PC on the `improve_camera_discovery`. That is I can continuously scan, connect, and reconnect without having to go into pairing mode on the camera. I added a simple test [file](https://github.com/gopro/OpenGoPro/blob/improve_camera_discovery/demos/python/sdk_wireless_camera_control/open_gopro/demos/reconnect_test.py) if you want to try it. If you get the same error, I'm going to be quite confused

tcamise-gpsw, 2022-08-26:
@ladyeklipse I pushed up a similar change for the tutorials on the `improve_camera_discovery` branch. Can you test this out? I've also added a reconnect test [file ](https://github.com/gopro/OpenGoPro/blob/improve_camera_discovery/demos/python/tutorial/reconnect_test.py)there to try to isolate the issue.

tcamise-gpsw, 2022-09-15:
I have been testing with this change and it appears to have fixed the problem. It is merged to main and has been released to pypi as of 0.11.0.

closed 2022-09-15 by tcamise-gpsw