gopro/OpenGoPro #221: need milliseconds in set time

steelbytes, 2022-08-21:
need milliseconds in set time.  and should really also have timezone.

I need this via bluetoothle, not via the qr code.

KonradIT, 2022-08-22:
Bluetooth has delay in sending the command and receiving it though, wouldn't be very accurate and make having millisecond level precision useless.

steelbytes, 2022-08-23:
> Bluetooth has delay in sending the command and receiving it though, wouldn't be very accurate and make having millisecond level precision useless.

yes.  but it's still better than the current api which only allows 1sec precision

tcamise-gpsw, 2022-08-24:
timezone and daylight / savings time support is coming soon.

but I agree with Konrad that, with the current BLE support, millisecond timing is pointless. It is very unlikely that this will ever be supported. However, I will leave this open for tracking

steelbytes, 2022-08-27:
I jsut read info on the GPSU metadata and that is also only to 1sec.

really really need millisecs.  yes I know it won't be perfect but 1sec is jsut utterly useless.

steelbytes, 2022-09-14:
> timezone and daylight / savings time support is coming soon.

I see in your docs that you now list a new version of get/set time that includes timezone+dst **BUT** you only list it for the hero11?  any chance of adding this to the firmware for the 10/9?

tcamise-gpsw, 2022-09-15:
Hi. There are no current plans for new 10 / 9 FW releases. If it were to happen, yes this would be included.

steelbytes, 2022-09-18:
> Hi. There are no current plans for new 10 / 9 FW releases. If it were to happen, yes this would be included.

why?  I would have thought it wouldn't be hard to add this to 9/10 ... ?