gopro/OpenGoPro #222: Bug Description

MDThomsen, 2022-08-24:
**Component**
- Documentation

**Describe the bug**
I'm trying to pair a HERO9 Black on android without using the GoPro Quik app.

The documentation for BLE states for pairing with either camera in step 4 you need to:
**Finish pairing with the peripheral**

What is meant by this? 
If you put the camera into pairing mode as described in the documentation and you pair with the camera with an android device without using the GoPro Quik app, then the camera will be shown as paired on the android device's bluetooth settings, but the GoPro camera will still be the pairing mode. There should be a way to tell the camera that pairing is completed, such that the phone is whitelisted?

**To Reproduce**
Steps to reproduce the behavior:
1. Set camera into pairing mode
2. Pair android device with camera not using the "GoPro Quik" app.
3. Camera is still in pairing mode.

**Expected behavior**
I would expect that the camera shows "Pairing successful" after explicitly clicking the "Pair" button that pop ups on android when attempting to pair with the camera.

**Hardware**
- Camera: Hero9 Black
- Phone: OnePlus 8 (Android 12)

tcamise-gpsw, 2022-08-24:
Hello. I agree that what you are describing is the correct behavior. That is, you should be able to pair from the Android OS without Quik.

And in fact,this is the behavior I see with my Gakaxy S20+ (Android 12) and Hero 10. So I can not replicate your problem.

Furthermore, Hero 8 is not [supported ](https://gopro.github.io/OpenGoPro/ble_2_0#supported-cameras) by Open GoPro.
So I'll leave this open for now in case you want to try replicating with a supported camera but this is likely specific to the older cameras, in which case it will be rejected.

tcamise-gpsw, 2022-09-15:
Closing due to inactivity and above use of unsupported camera

closed 2022-09-15 by tcamise-gpsw