gopro/OpenGoPro #224: Python API not working on ubuntu 20.04

carfreak4me, 2022-08-27:
unable to use the python API when installed from pip package manager (sudo pip install open-gopro)

reoccurring error is the following  

File "/usr/local/lib/python3.8/dist-packages/open_gopro/proto/turbo_transfer_pb2.py", line 5, in <module>
from google.protobuf.internal import builder as _builder
ImportError: cannot import name 'builder' from 'google.protobuf.internal' 

I have tried the first 2 pages of solutions on google without luck. 

any help would be appreciated

tcamise-gpsw, 2022-08-29:
Hello I just tried this in clean virtual environment on Ubuntu 20.04 and it works with no problems.

My guess is that you have conflicting versions of protobuf installed since you are using your system python. Firstly, it is not a good idea to be using sudo with pip. Beyond the safety [reasons](https://pages.charlesreid1.com/dont-sudo-pip/), you also run into dependency conflicts.

My recommendation is to use a clean [virtual environment](https://realpython.com/python-virtual-environments-a-primer/).

As a side note, I will soon be dropping support for Python 3.8 so you may want to move to 3.9 / 3.10.

tcamise-gpsw, 2022-09-15:
Any updates here?

closed 2022-11-15 by tcamise-gpsw