gopro/OpenGoPro #227: Remove Internet dependency when connecting to AP

lanfker, 2022-09-02:
**Is your feature request related to a problem or use case? Please describe.**
When I want to use my GoPro to do RTMP streaming, and if my home WiFi does not have Internet access, the Quik app can't ask GoPro to connect to my WiFi

**Describe the solution you'd like**
Want the GoPro cameras to be able to connect to Wi-Fi without Internet to conduct RTMP streaming

**Describe alternatives you've considered**
Cellphone hotspot, but I can't have my hotspot on always. 

**Additional context**
I was trying to do WLAN only video streaming, WLAN may not have Internet access.

tcamise-gpsw, 2022-09-15:
There is no such internet requirement for livestreaming. You can following the [livestreaming steps](https://gopro.github.io/OpenGoPro/http_2_0#downloading-preview-stream) and use whatever URL you want for the RTMP server.

Or are you specifically requesting that this is allowed from the Quik Mobile App? If so, that is out of the scope of Open GoPro.

tcamise-gpsw, 2022-11-15:
Closing due to inactivity and lack of problem

closed 2022-11-15 by tcamise-gpsw