gopro/OpenGoPro #229: Stream on custom Android/IOS app

m5a0r7, 2022-09-10:
Is the anyway to get access to a preview stream on an Android/IOS app like what we have in the quick gopro? 
I tried couple of approaches but in terms of quality and latency the quick go pro is completely different. I was wondering if anyone else has explored this feature and has a good solution for it. 

Thanks!

tcamise-gpsw, 2022-09-15:
Hello. Have you tried the [Preview Stream](https://gopro.github.io/OpenGoPro/http_2_0#downloading-preview-stream) from the spec?
If so, what exactly is the problem? Is it the latency?

m5a0r7, 2022-09-19:
Hi,

Thanks for the reply. Yeah I tried. The problem is that when I read the UDP packages from the 8554 port after starting the preview, I get choppy frames with a bit delay. 
I'm using the HERO 10 camera. I also noticed that the latency and quality is dependent on the camera mode, like vide, time lapse and photo. Maybe the encoding is different in each case. The video compression encoding on gopro is H.264 + HEVC. 
After all, I used the GStreamer to parse the UDP packages and after lots of trial and errors for a good setting, I could get a real time/high quality stream on the mobile device. I still get some choppy frames but the Quick app is working fantastic. I was wondering what is the best practice approach for streaming from a GoPro camera on an Android/IOS app, like what Quick go pro does. 

Looking forward to hearing from you.

tcamise-gpsw, 2022-09-20:
The short and unhelpful answer is that Quik is proprietary software which has been tweaked over many years for optimal streaming. So there could be any number of problems / inconsistencies in the decoder / library you are using.

However, I think we need to provide, at minimum, some "best practices" documentation for streaming. And also a demo that will show ideal streaming. I will update the ticket title to reflect this.

tangjw, 2022-11-17:
try ijkplayer, work fine on Android