gopro/OpenGoPro #235: Starting live stream with python package v0.11.0

kobezf, 2022-09-19:
I just installed the latest 0.11.0 version of the python gopro package and got an error 
```
  File "/usr/lib/python3.8/site-packages/open_gopro/__init__.py", line 14, in <module>
    from open_gopro.gopro import GoPro
  File "/usr/lib/python3.8/site-packages/open_gopro/gopro.py", line 22, in <module>
    from open_gopro.ble import BleUUID
  File "/usr/lib/python3.8/site-packages/open_gopro/ble/__init__.py", line 8, in <module>
    from .controller import BleDevice, BleHandle, NotiHandlerType, DisconnectHandlerType, BLEController
  File "/usr/lib/python3.8/site-packages/open_gopro/ble/controller.py", line 20, in <module>
    class BLEController(ABC, Generic[BleDevice, BleHandle]):
  File "/usr/lib/python3.8/site-packages/open_gopro/ble/controller.py", line 51, in BLEController
    def scan(self, token: Pattern, timeout: int = 5, service_uuids: list[BleUUID] = None) -> BleDevice:
TypeError: 'type' object is not subscriptable
```
I tried video.py from demos.

OS: Ubuntu 20.04

tcamise-gpsw, 2022-09-19:
What Python are you using? As the docs, state you must be using >= 3.9

I guess I should put a check for this in the code.

kobezf, 2022-09-23:
@tcamise-gpsw Sorry for delay, no, it was 3.8. I didn't notice these changes, sorry

closed 2022-09-28 by tcamise-gpsw