gopro/OpenGoPro #238: More IP camera like API

mms-, 2022-09-20:
While controlling functions is nice. This in the end is a camera. Grabbing live video or a live image should be a single http request. Not writing c++ code to read a UDP socket! Just run a RTSP server on camera like any IP camera does. Then any software can connect to the camera and it truly is 'open'. Too convoluted right now for simple scenarios.

tranzmatt, 2023-04-11:
This would be a wonderful feature.  I'm trying to stream from my two Hero 10's on the same desktop box and it's a nightmare.

tcamise-gpsw, 2023-12-08:
Should I keep this ticket open? Is there anything from the COHN solution that does not work for your use case?

mms-, 2023-12-10:
Hi, this is good news, have not had a chance to try this but it should do it.

As long as the rtsp is stable, easy for us or the end user to provision, supports 4K along with different view modes it will work nicely.

closed 2023-12-10 by mms-

tranzmatt, 2023-12-26:
Just saw this.  I assume I need to upgrade firmware?

tcamise-gpsw, 2024-01-08:
Yeah you'll need the latest firmware on Hero 12.

gtirth, 2025-05-07:
I have got this latest Hero Pro 13 camera and I am trying to start the rtsp streaming on it but I am not getting anything?

Can someone plesae help?