gopro/OpenGoPro #240: OpenGoPro API accessibily through station mode (STA)

d-biagini, 2022-09-26:
Hi everyone,
at my company we are building a remote controllable, multi-camera portable rig setup for a project we are developing from a while. We found in GoPro Hero series a very good choice for the cameras and we developed some concepts and prototypes. We are actually using GoPro Hero 10 cameras.

We implemented and tested large parts of the OpenGoPro API, and the only big and a bit frustrating limitation we found is the need, in the context of BLE + WiFi control method, to put the camera in AP mode and use a static, not changeable IP address.

This obliged us to take in some additional (not desidable) hardware to workaround the limitation and correctly operate our 5-cameras setup, so we were very pleased when we noticed that recently the API was updated to support STA mode for the GoPro Hero (I think to directly support Live Streaming).

I built a test and I could correcty put the camera in station mode and get a `NotififProvisioningState` notification indicating `PROVISIONING_SUCCESS_NEW_AP`.

Altough I can ICMP ping the camera after the event, it seems the API is not reachable through the WiFi interface while in this mode (tested with  `http://<camera-ip-address>/gopro/camera/shutter/start` , resulting in a timeout), and the interface seems to be pingable for a minute or two, before turning into unreachable. 
I may add that even when turning unreachable the provision status (status ID: 24) doesn't change and notifies `4: Completed`, as reported in API documentation.

I think that STA mode was brought in the camera firmware to correctly support Live Streaming feature and is not intended for camera control at the moment, but  it would be really, really nice to extend the support to OpenGoPro API and to bring in a way to seemleasly mantain the camera in STA mode while operating.

I hope I'll here something from you soon, have a nice day.

tcamise-gpsw, 2022-09-28:
Hello. Your understanding is correct. Wifi HTTP communication is not available in STA mode. This should have been in the spec, I will leave this open until it is added.

d-biagini, 2022-09-30:
Thank you @tcamise-gpsw for the clarification. We really hope to see this feature in an upcoming firmware version.

tcamise-gpsw, 2022-11-15:
FYI documentation has been updated. So this ticket is now solely to track the firmware request

ElD4n1, 2023-05-05:
Hi all.

Our company is also currently evaluating if we could use the GoPro for our needs. We are building a two-camera-system that should be controllable by a single smartphone. Our plan is that the phone instructs the cameras via BLE to connect to a common WiFi AP such that the phone is then able to take video recordings and download them from both cameras simultaneosly.

Currently it seems that this scenario is not possible (yet) due to the limitation described within this issue.

However, I found a place in the documentation which suggests that HTTP command & control is already possible on some cameras (the last sentence):

> The camera supports connecting to access points in [Station Mode (STA)](https://en.wikipedia.org/wiki/Station_(networking)). This is necessary for features such as Live Streaming, where the camera needs an Internet connection. **While in this mode, HTTP command and control of the camera is not available on some cameras.**

Source: https://gopro.github.io/OpenGoPro/ble_2_0#interface-with-access-points

Can you please elaborate a bit more what is already possible in STA mode regarding HTTP command & control and on which cameras?

pavlosharhan2, 2023-05-23:
Our company is interested in utilizing the HTTP API in STA mode. It would be awesome to have some Labs firmware that supports it.

@ElD4n1 Have you got any updates whether It's possible at least in some cameras? Thanks

ElD4n1, 2023-05-24:
> @ElD4n1 Have you got any updates whether It's possible at least in some cameras? Thanks

Unfortunately no, we found that the GoPro API currently has too many limitations for our use case (in addition to this issue) so I didn't investigate further.

tcamise-gpsw, 2023-12-06:
Hello. This should be solved now via the [Camera on the Home Network](https://gopro.github.io/OpenGoPro/http_2_0#camera-on-the-home-network-cohn) feature

closed 2023-12-08 by tcamise-gpsw