gopro/OpenGoPro #253: KeepAlive does not work on Hero11

EddyCelis, 2022-10-29:
Hero11 is set to auto power off after 5 minutes.
App is sending a keep alive packet to the camera every 30 seconds, which is ack'd by the camera (0x02 0x5B 0x00).
But camera powers off after 5 minutes.
No problem with Hero 7, 8, 9, 10.

tcamise-gpsw, 2022-11-15:
Hi. There is different keep alive logic in > Hero 10 which is now [documented](https://gopro.github.io/OpenGoPro/ble_2_0#keep-alive).

Can you try this with the recommended keep alive interval of 3 seconds?

EddyCelis, 2022-11-19:
OK, it works. A 2 second keepalive interval keeps the camera from going to sleep.

- From the documentation, it looks like keep-alive packets will have to be send also while the camera is recording. While the camera is recording, nothing is being send to the camera, but it receives a lot of status packets. Is the camera resetting the power down timer when it is SENDING something? I'm pretty sure the Hero 8, 9, 10 firmware did not require keepalive packets while recording.
- We are running a mobile app which is not time critical. Our 3 second interval might be stretched a bit every now and then. So, we have to shorten the interval to 2 seconds or 2.5 seconds. This makes for a pretty busy 'idle' connection. 
- Are future firmware updates for Hero 8,9,10 also going to require this shortened interval?
- Does this have a negative impact on battery life when connected to an idle camera?

tcamise-gpsw, 2023-01-03:
No, you do not need to send the keep alive while recording. I'll update the spec to make this clear.
I can't say definitively but it is very unlikely that future firmware updates for Hero <= 10 will require this (assuming such updates ever even happen).
I'm not exactly sure what you're referring to in regard to "idle" here but no, the battery impact should be negligible. I just took a sniffer capture to look at the keep alive message and the entire sequence takes about 500 us.

closed 2023-02-23 by tcamise-gpsw