gopro/OpenGoPro #268: Feature Description

PengKunPROO, 2022-11-28:
**Is your feature request related to a problem or use case? Please describe.**
I just wondering if it's possible to connect two or more goPros simultaneously and take those camera as two webcams!So that I could integrate those cameras in ROS and get their video stream at the mean time!

**Describe the solution you'd like**
Could you take another way for data transport?Cause I think there is no way to connect two cameras' wifi at the same time and using HTTP to get their data respectively and simultaneously! If it had,please tell me how to achieve it! Thanks!
Waiting for your reply!

tcamise-gpsw, 2022-11-28:
Hello. There is not currently any support for this on the GoPro side. 

There is work being done to allow multiple simultaneous webcams that should be released with the next firmware release. I will keep this ticket open now to track this, and potentially later if you need this for the Preview Stream also.

KonradIT, 2022-11-29:
How are the `multiple simultaneous webcams` streams done? Different port for each camera? I've been trying to get multiple UDP streams from 4x HERO9 for quite some time, and what ends up happening is my machine will mix all frames from all 4x UDP 8554 streams into each stream.

tcamise-gpsw, 2022-11-29:
Yep that's correct: different ports for each camera. You will be able to configure the port via an argument to the start webcam endpoint.

KonradIT, 2022-12-15:
Works well on latest HERO11 `v02.01` firmware:

`curl.exe "http://172.23.120.51/gopro/webcam/start?res=12&fov=0&port=8082"`

mehrajagdish, 2023-01-04:
Any updates on this i need to get live feed from two gopro's simultaneously connected via usb.

KonradIT, 2023-01-04:
Yup, use the `port` parameter, specify different ports for each.

mehrajagdish, 2023-01-04:
> Yup, use the `port` parameter, specify different ports for each. 
Can you provide any example regarding this,thanks.

tcamise-gpsw, 2023-01-04:
I'm working on updating to docs to add this parameter as well as adding a small python example for multiple webcams. I'll keep this open until these are released.

mehrajagdish, 2023-01-05:
> I'm working on updating the docs to add this parameter as well as adding a small python example for multiple webcams. I'll keep this open until these are released.

When multiple gopro's are connected via USB only one gopro is detected as a webcam at a time.
I used opencv capture function to capture multiple go pro's but i can get feed from only one device at a time on Windows.
And there is no option to directly get feed from gopro camera's over USB as go pro is not shown as webcam device in linux.
Any updates for GoPro as webcam on Linux devices over USB not via IP i need mininum latency and via IP the latency is high?

mehrajagdish, 2023-01-05:
> Works well on latest HERO11 `v02.01` firmware:
> 
> `curl.exe "http://172.23.120.51/gopro/webcam/start?res=12&fov=0&port=8082"`

not working on my hero11 v02.01 the port remains 8080 even after providing the port argument.

tcamise-gpsw, 2023-01-05:
This has been merged and you can find the demo [here](https://github.com/gopro/OpenGoPro/tree/main/demos/python/multi_webcam).

I don't really know what you're referring to because the default is not 8080; it is 8554.

In any case, the demos work or you can just do this in curl:

```
curl -v "http://172.29.192.51:8080/gopro/camera/control/wired_usb?p=0"
curl -v "http://172.29.192.51:8080/gopro/webcam/start?res=12&fov=0&port=8000"
```

You can then view from VLC, for example, at `udp://@0.0.0.0:8000`

closed 2023-01-05 by tcamise-gpsw

KonradIT, 2023-01-05:
Don't use the GoPro Webcam software. Ingest the UDP stream. No other way.

mehrajagdish, 2023-01-06:
> Don't use the GoPro Webcam software. Ingest the UDP stream. No other way.

I am able to start UDP stream for both gopro's at different port but while reading frames from both streams using opencv videocapture read method i am getting lag in one of the UDP streams and both the streams are out of sync . How can i reduce the lag and get both the streams in sync.

KonradIT, 2023-01-06:
Check if the UDP packets contain a timecode, and adjust accordingly.

mehrajagdish, 2023-01-06:
> Check if the UDP packets contain a timecode, and adjust accordingly.

Solved by using multiprocessing,Thanks