gopro/OpenGoPro #271: Pairing without a Discovery request

panekk, 2022-12-05:
**Is your feature request related to a problem or use case? Please describe.**
I'm trying to pair GoPro with a Garmin watch, which unfortunately it's not capable of discovering available services after pairing. However, I can say that on the BT layer GoPro is for sure paired, as pairing callback from Garmin SDK is called and I can read characteristics available on GoPro. It doesn't do anything after sending commands, though, and I suppose it's because it never goes of connection screen (I never see "Connection successful", and after exiting this screen, the communication still works, but nothing happens after write).

**Describe the solution you'd like**
I'd like GoPro to be paired and in operable state when BT pairing succeeds, without discovering the services.

**Describe alternatives you've considered**
Unfortunately I don't see any for this specific use case.

tcamise-gpsw, 2022-12-05:
Hello. There is no dependency on the GoPro side regarding discovering services. The only reason you need to discover services is so that you know what handles to write to enable the relevant characteristic notifications. So you can discover characteristics before pairing if you desire.

What exactly are you you trying to do that is failing? What command are you sending and what are you receiving back? Can you post a Bluetooth data log or something similar?

panekk, 2022-12-06:
Hello. Thank you for the quick response.

Unfortunately I'm not able to get detailed logs, at least right now (I'll try tomorrow, though), so for now I'll describe the procedure I'm doing.
1. On GoPro: Preferences -> Wireless Connections -> Connect Device -> GoPro Quik App.
2. On watch/simulator: Scan for GoPro.
3. On watch/simulator: Initiate pairing and wait for success.
4. On watch/simulator: Read Battery level (00002a19-0000-1000-8000-00805f9b34fb), value is correct.

This works, so I suppose the pairing is complete, but on GoPro I'm still in the "GoPro Quik App" screen, without getting "Connection successful". I tested the same approach with my phone, Windows PC and provided Python script from tutorial 1 (with disabled notifications to make sure it's not related), and everywhere there's "Connection successful" message, so my guess was that for some reason GoPro expects to have services discovered by the client after pairing.

However, during writing this post I made more tests and have new findings. nRF Connect actually can pair without discovering services. It also can read Battery level without pairing, so it doesn't prove anything that I can do it. However, b5f90002-aa8d-11e3-9046-0002a5d5c51b (camera name) can be read only after pairing, and this read fails in my case. Most likely I ran into a bug in Garmin's SDK saying me the device is paired, when it's not.

closed 2022-12-13 by tcamise-gpsw

George-Gi, 2024-07-08:
Hi, I am trying to control my GoPro through a Garmin app that I am developing and I have a similar issue. I can connect and read the characteristics, but when I send commands nothing happens. I get back an error regarding authentication. The API for my watch doesn't support bonding unfortunately. Do I strictly have to request a bond, or I can send commands without it?
@panekk Did you have any luck? I am also using the nrf dongle and I am short of hitting a dead end.

tcamise-gpsw, 2024-07-08:
Hello. To clarify, you can not use the Open GoPro API without pairing / bonding. If the SDK you are using has no way of doing this, then that is a problem with that SDK. If you are able to pair and still seeing issues feel free to open an issue here.

tcamise-gpsw, 2024-07-08:
The "Open GoPro API" is the communication with the GoPro. So you are using the Open GoPro API. 

To be extremely explicit: no, you can not communicate with the GoPro without pairing / bonding.