gopro/OpenGoPro #272: BLE device name wrong

csscriptplus, 2022-12-07:
**Component**
What is the bug in?
- Demos (ionic)
- Specs (BLE)

**Describe the bug**
I have reset my GoPro to factory settings. The GoPro then has a default device name. When the GoPro demo app searches for bluetooth devices, it finds the GoPro. But in the SearchResult object device name is not set. The app shows "NO Name". After 2nd scan device name is set with "GoPro 1996". But this is not the default device name. If I then change the device name via the GoPro Quick App, then the GoPro also shows this name in the settings. But the BLE scan in the demo app still returns "GoPro 1996".
Where does this device name come from? Why is not the set device name displayed? Why does this not work on the first scan?

**Hardware**
- Camera: GoPro10
- iPhone SE

tcamise-gpsw, 2023-12-06:
Which demo app are you referring to?

tcamise-gpsw, 2024-01-08:
Ok well that app was provided by a community member so I can't comment on it. But I'll try to set up a similar tests to reproduce what you are seeing.