gopro/OpenGoPro #286: List index out of range error when using more than one WiFi

hhackbarth, 2023-01-04:
**Component**
Python SDK

**Describe the bug**
When a second WiFi adapter is attached to the system (in my case Ubuntu 22.04 on RasPi), the procedure to discover the desired WiFi AP terminates with
Error while opening: list index out of range (gopro_wireless.py:326)
This even happens, when the adapter to be used, is given as command line parameter.

**To Reproduce**
Prerequisite: You need two activated WiFi network adapters (the built-in as 'wlan0' and another one attached via USB).
Steps to reproduce the behavior:
1. Start gopro-wifi using 'gopro-wifi --wifi_interface wlan0'
2. See error "IndexError: list index out of range"

**Expected behavior**
System should be connected via WiFi adapter wlan0 to GoPro AP.

**Hardware**
- Camera: Hero11 Black Mini
- Desktop: Ubuntu MATE 22.04 64bit (arm64)
- Raspberry Pi 4B

**Additional context**
Current workaround is to unplug the USB WiFi adapter, start the gopro-wifi procedure and after successful connection, plug in the USB WiFi again.
[index_out_of_range_error.txt](https://github.com/gopro/OpenGoPro/files/10345097/index_out_of_range_error.txt)

hhackbarth, 2023-01-04:
Thanks for your quick answer. It is a problem with multiple adapters. The log only shows that it is searching for the SSID retrieved via BLE but during this search procedure it terminates, when there are multiple adapters.

I attached two logs taken with logging level DEBUG for util.py. The logfile "index_out_of_range.txt" shows the output, when two wifi network adapters are active. The other file "no_error.txt" shows the log output, when only one adapter (wlan0) is active. In that case everything goes fine and it proves that it is a problem with multiple wifi adapters.

I think it would be much easier for you to trace what happens, if you attach a second (USB) WiFi adapter to your test system as well.

I definitely would prefer to use only one adapter and let the GoPro work in station mode (as wiFi client). It would be much easier to transcode and provide the stream to my other systems as desired, but as this is currently not supported by GoPro 11, we have to live with multiple adapters.

[index_out_of_range_error.txt](https://github.com/gopro/OpenGoPro/files/10347763/index_out_of_range_error.txt)
[no_error.txt](https://github.com/gopro/OpenGoPro/files/10347764/no_error.txt)

tcamise-gpsw, 2023-01-05:
The error log is not actually showing the debug logging. I can't say why this...maybe you didn't install the package in editable mode. Anyway, I will try to replicate at some point on my ubuntu machine

tcamise-gpsw, 2023-12-08:
I do not have a setup to test this functionally. I'll try writing unit tests with multiple adapters to see if I can replicate.