gopro/OpenGoPro #289: Bug on webcam_exit Unexpected error: ValueError('23 is not a valid MediaFormat')

quintene, 2023-01-05:
**Component**
What is the bug in?
- Demos (python sdk)

**Describe the bug**
When the webcam mode is closed by using the demo script it will return the following error: 
"Unexpected error: ValueError('23 is not a valid MediaFormat')". Stopping the camera will work with the stop command but the exit command crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the webcam demo script 

**Expected behavior**
Webcam to be stopped and webcam mode to be closed.

**Hardware**
- Camera: Hero9, etc.
- Desktop if applicable [e.g. Windows]
- Smartphone if applicable [e.g. iPhone6]

tcamise-gpsw, 2023-01-05:
Hi. Thanks for this. I will push a quick fix for this in the Python SDK to add this value to the MediaFormat enum. This will at least prevent the error from occurring.

But I'm also going to use this ticket to figure out why this value (and the other unknown values from the Enum) are not documented in the spec. They should be getting automatically documented and then added to the Python SDK.

tcamise-gpsw, 2023-01-05:
OK the placeholder fix has been merged to `main`

closed 2023-01-06 by quintene

tcamise-gpsw, 2023-01-12:
Just to finalize here...the presence of these unknown Media Format values was a firmware bug in certain GoPro firmware releases. It is not fixed, at least on Hero >= 10.

In order to handle cameras that still contain the bug, I have changed the Python SDK to only warn (and not break) when it receives unknown setting / status values.

Also i had to make a minor change in the webcam demo to support Hero 9 which was part of the linked PR.

In any case, this ticket is now complete.