gopro/OpenGoPro #300: GoPro Hero 10 Multi Webcam stream doesn't stream on a specified Port number

praneetheddu, 2023-01-23:
**Component**
Python Multiwebcam 

**Describe the bug**
This is in reference to this [issue](https://github.com/gopro/OpenGoPro/issues/268). Has the multi-webcam stream been tested for Go Pro Hero 10 (Firmware: v1.50)? Starting the webcam without specifying the port for a single GoPro works fine and I'm able to view the stream on VLC player or OpenCV video capture on the default port 8554 <br/>

However, when starting the webcam stream on a specific port, I'm unable to ingest the UDP stream on that  port. I've tried using the recommended command in the referenced issue (`curl -v "http://172.29.192.51:8080/gopro/webcam/start?res=12&fov=0&port=8000"`) to start the webcam. Instead, the stream is still shown on the default port 8554 which is undesired as it leads to corrupted frames when using multiple cameras. I went through the ReadMe tutorial described [here](https://github.com/gopro/OpenGoPro/tree/main/demos/python/multi_webcam) and still no luck. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect GoPro Hero 10 via USB on Ubuntu 22.04
2. Run `curl -v "http://172.29.192.51:8080/gopro/webcam/start?res=12&fov=0&port=8000"` (change the IP to reflect your camera's serial number) to start webcam on the specified port
3. Start VLC media player and open a network stream on `udp://@0.0.0.0:8000` (Alternatively, use OpenCV Video Capture)
4. Stream doesn't show and the loading screen hangs

**Expected behavior**
View the webcam stream when launched on a specific port

**Hardware**
- Camera: Hero10 Black.
- OS: Ubuntu 22.04 LTS

alecive, 2023-01-23:
Go through their support. If they have github, open an issue.

praneetheddu, 2023-02-13:
@afong3 Thank you for the response. Yes, I was using dual GoPro Hero 10 Black to run the multi-webcam service but ran into the same issues you mentioned. I'll go for the GoPro Hero 11 and start the stream on a specified port (Also, thank you for confirming that this feature works!)

tcamise-gpsw, 2023-02-23:
Hello all. Yes, that is correct that the port parameter is only supported on certain camera's / FW versions.

I think it is out of the scope of the demo to check / enforce this but I will at least add something to the demo documentation to note this.

tranzmatt, 2023-04-05:
So is there a workaround for HERO10's?  I have need of two GoPro 10's running on the same host and was beating my head on this a bit.

tcamise-gpsw, 2023-12-06:
Unfortunately no, there is no plan to back port this to older devices.

closed 2023-12-06 by tcamise-gpsw