gopro/OpenGoPro #301: wifi_interface

isdito, 2023-01-23:
Hello,

I Have this interface

```
Hay 1 interfaz en el sistema:

    Nombre                   : Wi-Fi
    Descripción            : TP-Link Wireless USB Adapter
    GUID                   : 2a588865-9192-44d3-9dfa-56c24084f2ec
    Dirección       : 98:48:27:b7:7a:74
    Tipo de interfaz         : Principal
    Estado                  : desconectado
    Estado de radio           : Hardware Activado
                             Software Activado

    Estado de la red hospedada: No disponible
```
I bought a gopro 11, to try to connect via bluetooth or wiffy. I want to be able to record a certain time and send something very simple to the computer that is taking me a long time to focus on what I have to do. On my other gopro5 it is very simple.

I don't know what the wiffy interface is and how I have to put the command.

gopro-wifi --wifi_interface Wi-Fi

this not works

reggards

isdito, 2023-01-24:
Hello, for the next that have the same problem than I, it is the language of the windows you must have in english because not fine the "nombre"  the wifi interface find "name"

tcamise-gpsw, 2023-02-23:
Yes, unfortunately the Wifi "driver" requires CLI response parsing and thus requires English as you have found.

I will add a run-time check to validate that the system language is US English.

closed 2023-02-24 by tcamise-gpsw

tcamise-gpsw, 2023-02-24:
I have merged and released (in v 0.13.0) the documentation updates for this