gopro/OpenGoPro #312: Python Demos import GoPro Error

Bartelime, 2023-02-11:
Dear all
I try to play with the python demos. After installing everything sucessfully, (Python 3.9.3) according 
https://gopro.github.io/OpenGoPro/python_sdk/troubleshooting.html
I tried to play with the demos.
I can not even import GoPro, it gives an error. 

from open_gopro import GoPro   --> ImportError: cannot import name 'GoPro' from 'open_gopro'

If I try to import the Params, this works.

I tried already several things (and days) (uninstall, new install, different python version.....). I would really appreciate someones help.

If I try to go ahead on the command line for example "gopro-photo" it seems to work, but ends up with some error message with the wifi (but my laptop is connected with the wifi from the GP10, FW 1.5)
--> Error while opening: Das Profil WIFINAME wird der Schnittstelle WLAN        gopro_wireless.py:326
         hinzugefgt.

Is there someone else with similar experience or am I doing something completely wrong?

Regards
Bartelime

tcamise-gpsw, 2023-02-23:
Hello. It looks like that snippet documentation you are pointing to is outdated and needs to be updated. There are now two distinct top level GoPro classes as documented in the usage: https://gopro.github.io/OpenGoPro/python_sdk/usage.html#opening

Thanks you for pointing this out as I realize now this occurs in several places. I will update the documentation then close this ticket.

closed 2023-02-24 by tcamise-gpsw

tcamise-gpsw, 2023-02-24:
I have merged and released (in v 0.13.0) the documentation updates for this