gopro/OpenGoPro #313: LIVE_STREAM_ERROR_CAMERA_BLOCKED error when not calling RequestConnect right before RequestSetLiveStreamMode

petrpavlik, 2023-02-13:
**Component**
What is the bug in?
- Demos (python sdk, etc)
- Tutorials
- Documentation
- Specs (BLE , HTTP)

**Describe the bug**
Trying to trigger a live stream using the BLE API without calling `RequestConnect` or `RequestConnectNew` before results in `LIVE_STREAM_ERROR_CAMERA_BLOCKED` error, even though the GoPro reports to be connected to a wifi network.

Please not that if we do call `RequestConnect` to the wifi the GoPro is reportedly already connected to right before calling `RequestSetLiveStreamMode`, everything works as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to a GoPro using the BLE API
2. Query that the camera is currently connected to a wifi using status id 29 (Provisioned WIFI AP SSID. On BLE connection, value is big-endian byte-encoded int)
3. Send `RequestSetLiveStreamMode`
4. See the list of status updates (oldest first)

```
> LiveStreamStatus(bitrate: 0, status: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamStatus.liveStreamStateConfig, error: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamError.liveStreamErrorNone)

> LiveStreamStatus(bitrate: 0, status: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamStatus.liveStreamStateReady, error: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamError.liveStreamErrorNone)

< Send request open shutter to the GoPRo

> LiveStreamStatus(bitrate: 0, status: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamStatus.liveStreamStateReady, error: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamError.liveStreamErrorCameraBlocked)

> LiveStreamStatus(bitrate: 0, status: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamStatus.liveStreamStateConfig, error: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamError.liveStreamErrorCameraBlocked)

> LiveStreamStatus(bitrate: 0, status: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamStatus.liveStreamStateConfig, error: PulseModel.OpenGoPro.LiveStreamStatus.LiveStreamError.liveStreamErrorNone)
```

**Expected behavior**
The camera starts live streaming

**Hardware**
- Camera: Hero11 Black, 02.01

Might be related to https://github.com/gopro/OpenGoPro/issues/259

tcamise-gpsw, 2023-02-23:
Hello. I don't disagree that what you are trying to do is reasonable.  However, anything other than the [exact documented sequence](https://gopro.github.io/OpenGoPro/ble_2_0#live-streaming-sequence) is not tested and thus not officially supported.

I will leave this open but don't expect this to change any time soon. It might help prioritization if you can provide some motivation as to the tradeoff / problems this additional step is causing you.

petrpavlik, 2023-02-24:
Thanks, I'm happy to close this if this flow is not supported.

Just wanted to ask first if substituting `RequestConnectNew` for `RequestConnect` is officially supported when setting up live streaming, provided the access point is provisioned. It seems to work for us for the most part, and another part of the docs states 
> Note: This should only be done once to provision the AP; subsequent connections should use [RequestConnect](https://gopro.github.io/OpenGoPro/ble_2_0#protobuf-commands).

Our motivation in general is to make live streaming through a GoPro a 1 button tap flow in an iOS app, so any shortcut we can potentially take is helpful.

mahboobkarimian, 2023-07-20:
I can confirm that this problem exists in Hero 9. With the latest API, if I use `RequestConnectNew` instead of `RequestConnect` for an already-known SSID, it is working fine. Otherwise no.

tcamise-gpsw, 2023-12-06:
I'm going to close this since:
- there is a documented working flow
- there is not enough reason to fix the broken flow

closed 2023-12-06 by tcamise-gpsw