gopro/OpenGoPro #314: Parse Error when running pip install for python tutorials

lsmith24, 2023-02-14:
**Component**
What is the bug in?
- Tutorials

**Describe the bug**
I am following the steps here https://gopro.github.io/OpenGoPro/tutorials/python/connect-ble to run the tutorials but I am getting this error:
`Obtaining file:///D:/Toolkit/OpenGoPro-2.0/demos/python/tutorial
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in open_gopro_python_tutorials setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Parse error at "'=0.12.1'": Expected stringEnd
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ..\demos\python\tutorial
2. Run "pip install -e ." 
3. see error

**Expected behavior**
It installs the tutorial materials.

**Hardware**
- Windows 11

tcamise-gpsw, 2023-02-14:
What Python version are you using? I just tested on 3.10 and it worked fine

lsmith24, 2023-02-14:
I'm using 3.9

tcamise-gpsw, 2023-02-15:
I just tested on 3.9.10 on Ubuntu and it worked fine. I don't currently have access to a Windows PC to test with but the tutorials were developed on Windows so it is very unlikely that this is OS-specific.

My best guess is your local pip / setuptools need to be updated, something like [this](https://github.com/sdispater/pendulum/issues/187#issuecomment-375820769).

Are you installing from a clean virtual environment? Or simply using your global Python environment?

GiscardBiamby, 2023-03-09:
I just tried with a brand new conda environment, using python 3.11 and the latest versions of pip and setuptools. Getting this error now. Here is the output from my environment build script:

```bash
ENV_NAME: sem_env, PYTHON_VERSION: 3.11

Remove all packages in environment /home/gbiamby/mambaforge/envs/sem_env:

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.3.0) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['python=3.11']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
Transaction

  Prefix: /home/gbiamby/mambaforge/envs/sem_env

  Updating specs:

   - python=3.11

  Package               Version  Build               Channel                    Size
──────────────────────────────────────────────────────────────────────────────────────
  Install:
──────────────────────────────────────────────────────────────────────────────────────

  + _libgcc_mutex           0.1  conda_forge         conda-forge/linux-64     Cached
  + _openmp_mutex           4.5  2_gnu               conda-forge/linux-64     Cached
  + bzip2                 1.0.8  h7f98852_4          conda-forge/linux-64     Cached
  + ca-certificates   2022.12.7  ha878542_0          conda-forge/linux-64     Cached
  + ld_impl_linux-64       2.40  h41732ed_0          conda-forge/linux-64     Cached
  + libffi                3.4.2  h7f98852_5          conda-forge/linux-64     Cached
  + libgcc-ng            12.2.0  h65d4601_19         conda-forge/linux-64     Cached
  + libgomp              12.2.0  h65d4601_19         conda-forge/linux-64     Cached
  + libnsl                2.0.0  h7f98852_0          conda-forge/linux-64     Cached
  + libsqlite            3.40.0  h753d276_0          conda-forge/linux-64     Cached
  + libuuid              2.32.1  h7f98852_1000       conda-forge/linux-64     Cached
  + libzlib              1.2.13  h166bdaf_4          conda-forge/linux-64     Cached
  + ncurses                 6.3  h27087fc_1          conda-forge/linux-64     Cached
  + openssl               3.0.8  h0b41bf4_0          conda-forge/linux-64     Cached
  + pip                  23.0.1  pyhd8ed1ab_0        conda-forge/noarch       Cached
  + python               3.11.0  he550d4f_1_cpython  conda-forge/linux-64     Cached
  + readline              8.1.2  h0f457ee_0          conda-forge/linux-64     Cached
  + setuptools           67.5.1  pyhd8ed1ab_0        conda-forge/noarch       Cached
  + tk                   8.6.12  h27826a3_0          conda-forge/linux-64     Cached
  + tzdata                2022g  h191b570_0          conda-forge/noarch       Cached
  + wheel                0.38.4  pyhd8ed1ab_0        conda-forge/noarch       Cached
  + xz                    5.2.6  h166bdaf_0          conda-forge/linux-64     Cached

  Summary:
[... 105 lines cut ...]
Collecting matplotlib
  Using cached matplotlib-3.7.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (11.6 MB)
Collecting nb-black
  Using cached nb_black-1.0.7.tar.gz (4.8 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [3 lines of output]
      error in nb_black setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after name and no valid version specifier)
          yapf >= '0.28'; python_version < '3.6'
               ^
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```

tcamise-gpsw, 2023-03-09:
The project does not currently support 3.11. Support for 3.11 will be added in the next few weeks.

closed 2023-12-06 by tcamise-gpsw