gopro/OpenGoPro #315: Live streaming can work without internet access

stress-ins, 2023-02-19:
**Is your feature request related to a problem or use case? Please describe.**
Use case -  I want live streaming to local RTMP in LOCAL wi-fi (without internet access), but I Can`t connect to wi-fi.

**Describe the solution you'd like**
I want go pro can connect and live streaming without internet access.

**Describe alternatives you've considered**
Add to wifi internet access, but when i turn off internet - go pro shutdown live stream and stop write video to SD

stress-ins, 2023-02-19:
#227  - maybe duplicated. But I try GoPro Quick APP and API call - it`s not work in step connect to WIFI access point.
And stop record video when connect to internet down.

mahboobkarimian, 2023-07-21:
I can confirm this statement with hero9.

> #227 - maybe duplicated. But I try GoPro Quick APP and API call - it`s not work in step connect to WIFI access point. And stop record video when connect to internet down.

mahboobkarimian, 2023-07-21:
Indeed it looks for api.gopro.com, and probably it does a TCP/TLS handshake.
I tried to put this address in hosts, but this didn't help.

soraxas, 2023-09-24:
Hi, I am experiencing the same thing. My GoPro Hero 11 can connect to a router that has WAN connection, but if I unplug WAN (and hence the router only has LAN), my GoPro refuse to connect to the router. 

This is true regardless of initiating the connection via the Android App, or via QR code command.

SergejJurecko, 2024-06-04:
Are there any plans to fix this? This prevents integrating GoPro with an android app in any kind of user friendly way. Since you can only programmatically start hotspots on android without internet.

tcamise-gpsw, 2024-09-10:
This is going to be supported in the next Hero 13 release.

tcamise-gpsw, 2025-03-25:
This should be fixed as of Hero 13 Black v 2.0.0

RexTech46, 2025-04-16:
I think it is not fixed. I try to connect to a local Wifi without an internet access. Using a Hero13 with labs firmware but it does not connect to the network using the app or QR codes.

tcamise-gpsw, 2025-04-23:
You're correct. There is actually a new protobuf property that needs to be set. I'll update the docs.

tcamise-gpsw, 2025-04-23:
Docs have been updated. There is a new [bypass_eula_check](https://gopro.github.io/OpenGoPro/ble/protocol/protobuf.html#requestconnectnew) field in the `RequestConnectNew` protobuf message that must set to True in order to allow connecting to a network without internet access.

closed 2025-04-23 by tcamise-gpsw

demiantres, 2025-04-24:
When I use ``bypass_eula_check`` then the Hero 13 (latest official firmware) refuses to go into live streaming mode (even if the router is connected to the internet).