gopro/OpenGoPro #317: Can a GoPro connect to an unsecured wifi (no password)?

petrpavlik, 2023-02-24:
The BLE protobuf specs state that `pasword` field for `RequestConnectNew` is required https://github.com/gopro/OpenGoPro/blob/main/protobuf/network_management.proto.

What we've tried to do is check the flags of returned `ScanEntry` and if `SCAN_FLAG_AUTHENTICATED` is not present, we try to call `RequestConnectNew` with an empty password. The result seems to be `PROVISIONING_ERROR_PASSWORD_AUTH` error.

Is connected a GoPro to a wifi that's not password-protected supported?

tcamise-gpsw, 2023-12-06:
Interestingly the answer is currently no. But this certain seems like a valid use case.

tcamise-gpsw, 2024-04-16:
After discussing internally this is the currently expected behavior. I'll leave this open but change to enhancement.