gopro/OpenGoPro #320: English language verification for Wifi Driver for C.UTF-8 as supported language

simohn, 2023-03-23:
**Problem**
I would like to use the latest version of the Python SDK on Ubuntu Core.
The problem is, that the library requires en_US.UTF-8 as locale, but on Ubuntu Core the locale is C.UTF-8 and unfortunately it's not editable. The language check was introduced in version 0.13.0.
https://gopro.github.io/OpenGoPro/python_sdk/changelog.html

**Describe the solution you'd like**
Can you add C.UTF-8 support?

tcamise-gpsw, 2023-03-23:
Hello. Sure we can add this. Can you post the error message you are getting so I know exactly what string to compare against?
It shoud be something like:
```
The Wifi driver parses CLI responses and only supports en_US where your language is {language}
```

tcamise-gpsw, 2023-03-23:
Although now that I'm reading about what C.UTF-8 [actually is](https://stackoverflow.com/questions/55673886/what-is-the-difference-between-c-utf-8-and-en-us-utf-8-locales) I am worried that it is not going to work. If the parsing responses are different in any way than en-US, it is very likely to break the parsing.

Anyway when you send me the error string, I will make a branch that accepts it but I'll leave it up to you to test.

simohn, 2023-03-23:
> Hello. Sure we can add this. Can you post the error message you are getting so I know exactly what string to compare against? It shoud be something like:
> 
> ```
> The Wifi driver parses CLI responses and only supports en_US where your language is {language}
> ```

Hello. Yes, here it is:

Traceback (most recent call last):
  File "/snap/cam-controller/x7/bin/cam-controller", line 8, in <module>
    sys.exit(main())
  File "/snap/cam-controller/x7/lib/python3.10/site-packages/cam_controller/app.py", line 5, in main
    with WirelessGoPro() as gopro:
  File "/snap/cam-controller/x7/lib/python3.10/site-packages/open_gopro/gopro_wireless.py", line 166, in __init__
    wifi_controller=wifi_adapter(wifi_interface, password=sudo_password) if enable_wifi else None,
  File "/snap/cam-controller/x7/lib/python3.10/site-packages/open_gopro/wifi/adapters/wireless.py", line 86, in __init__
    ensure_us_english()
  File "/snap/cam-controller/x7/lib/python3.10/site-packages/open_gopro/wifi/adapters/wireless.py", line 44, in ensure_us_english
    raise RuntimeError(
RuntimeError: The Wifi driver parses CLI responses and only supports en_US where your language is C.UTF-8

simohn, 2023-03-23:
> Although now that I'm reading about what C.UTF-8 [actually is](https://stackoverflow.com/questions/55673886/what-is-the-difference-between-c-utf-8-and-en-us-utf-8-locales) I am worried that it is not going to work. If the CLI responses are different in any way than en-US, it is very likely to break the parsing.
> 
> Anyway when you send me the error string, I will make a branch that accepts it but I'll leave it up to you to test.
> 
> Reading further it sounds like I should in fact be setting the language to C and parsing that instead of us-EN. If that's the case we can leave this open as a feature request but I have no idea when I'll get to that.

Ok, I understand. Yes, that sounds also reasonable for me.
Hmm. I think I will stay by an older version then and switch to the new release, in case you are going to change it.
Thank you.

simohn, 2023-04-04:
Is it possible to access the documentation of an older version?
On https://gopro.github.io/OpenGoPro/python_sdk/index.html I only get to the latest.

Creepios, 2023-07-28:
You can fix this temporarily with this:
```
import os
os.environ["LANG"] = "en_US"`
```

MelVimL, 2024-05-14:
If its okay I can try to add Windows with locale de_DE. Maybe some Linux Controller with de_DE too. Document the process to make it others easier to add their locals.

tcamise-gpsw, 2024-05-14:
Sure go for it! Please make a pull request here when its reqdy. 

Also, I will at some point refactor this to make it easier to add other languages.

MelVimL, 2024-05-14:
Should I edit it in the file or is it okay if I split it up into OS adapter files like:

[image]

zeyad-shaban, 2024-07-13:
run in temrianl `export LANG=en_US.UTF-8`

martincarapia, 2025-03-21:
Just clicking around, it seemed like this was super close to being finished. What's the current progress?

MelVimL, 2025-03-21:
I decoupled the windows driver and need to test it. After that I want to add the Linux. Last Year my freetime shrinked to a minimum and the gopro related project i worked for is finished. I think next time off I can finish it.