gopro/OpenGoPro #323: GoPro Hero 11 - Naming files

popovicidinu, 2023-03-30:
Hi everyone, 

I've just started working with a GoPro camera and the API, so I'm rather new to the whole ecosystem so apologies if my question is stupid. 

Looking at the documentation I couldn't find (or don't know if it's possible) to name a file (recording) that will later on be saved on the SD card. 

My use case is as follows: 

1. I want to start to start a video recording via the API but predefine the name of the final video file that gets saved. 
2. Once recording is stopped - the file is saved on the SD card. 
3. I can later on retrieve the file based on the name that I provided. 

I found some ideas around with using the last file name, or something like this - but not sure if this is an option? (e.g. what if the file is larger and takes time to be saved? How do I really know its the last file?). 

Note: I am trying to achieve this via a GoPro connected on USB.

KonradIT, 2023-03-30:
Not possible.

popovicidinu, 2023-03-30:
What about recording a video and then calling the getMediaList to retrieve the last file created? And base the logic on that?

KonradIT, 2023-03-30:
Sure. That tracks.

popovicidinu, 2023-03-31:
@KonradIT thank you for the info. I will work around that logic.

closed 2023-03-31 by popovicidinu