gopro/OpenGoPro #331: WiredGoPro is not correctly closed

Laurendus, 2023-04-21:
**Component**
What is the bug in?
python sdk

**Describe the bug**
The `close()`-Method of the WiredGoPro appears to be a stub which does not correctly reset the state of a GoPro connected via USB.

**To Reproduce**
1. Using the context manager successfully connect via USB to the GoPro.
2. Leave the context manager, triggering a call to `close()`.
3. Try connecting again. This will fail since the camera remains in a connected state.

**Expected behavior**
Closing the context manager successfully disables the usb control to support repeatedly entering the context manager without having to physically disconnect the camera to reset its state.

**Screenshots**
-/-

**Hardware**
- Camera: HERO10 Black
- macOS

**Additional context**
The following code appears to at least mitigate the problem. I am unsure whether additional cleanup is required.

```python
    def close(self) -> None:
        """Gracefully close the GoPro Client connection"""
        self.http_command.wired_usb_control(control=Params.Toggle.DISABLE)
        self._open = False
```

Ganarloo, 2023-04-25:
Hi Laurendus, I'm having an issue with connecting to my GoPro 10 via USB (https://github.com/gopro/OpenGoPro/issues/330). I'm very curios to know what code you're using to connect to the GoPro. Would you be able to share it?

tcamise-gpsw, 2023-12-06:
Yeah this unfortunately has not been tested yet.

tcamise-gpsw, 2024-04-10:
Closing this since I've pinned a known issue about the need for a major investigation into Python SDK BLE connection robustness [here](https://github.com/gopro/OpenGoPro/issues/487)

closed 2024-04-10 by tcamise-gpsw