gopro/OpenGoPro #333: Can not take a photo while in webcam mode but goes well in live stream mode (GoPro 11 black)

leongthb, 2023-04-23:
**Component**
What is the bug in?
- tutorial_modules of python
- Specs (HTTP)

**Describe the bug**
Hello, I am using GoPro Open API tutorial modules of Python and using wifi method to conrtol GoPro 11 black.
The problem is:
When starting the webcam: /gopro/webcam/start,  I found it can not taking a photo or recording a video by python at the same time.
It goes well when I Start preview stream, I mean when I use /gopro/camera/stream/start, I can take a photo or record a video at the same time.
I would appreaciate it very much if anyone can any advice on this, thank you very much!

**To Reproduce**
Steps to reproduce the behavior:
1. start webcam: eg, url = GOPRO_BASE_URL + "/gopro/webcam/start"
                     response = requests.get(url)
2. take a photo, eg, await client.write_gatt_char(COMMAND_REQ_UUID, bytearray([3, 1, 1, 1]), response=True)
3. Found error: When starting the webcam,  I found it can not taking a photo or recording a video by python at the same time.

**Expected behavior**
I need to display the live screen for multiple GoPro cameras 11 black and control them to take phots or record a video at the same time.

**Screenshots**
N/A

**Hardware**
- Camera: Hero11 black, etc.
- Ubuntu 22.04.1 X64

KonradIT, 2023-04-23:
Yeah this is how it works, once you put the GoPro in webcam mode you can't do anything else but to consume the stream. Exit out of webcam mode and take the photo/record a video.

tcamise-gpsw, 2023-12-06:
I'm going to close this since the original description is expected behavior as mentioned above.

To answer your most recent question...firstly I assume you are talking about **Preview** stream and not **Live** stream (i.e. RTLS) since the `gopro/stream/start` endpoint is for Preview stream.  If so, this is not currently possible. However, the next firmware release is going to add a `port` a parameter to the Preview Stream endpoint similar to what has been recently done for Webcam streaming. So this will soon be possible.

closed 2023-12-06 by tcamise-gpsw