gopro/OpenGoPro #337: Query the GoPro's IP address via BLE when in STA mode

ElD4n1, 2023-05-05:
**Is your feature request related to a problem or use case? Please describe.**
As a successor to #240, we would need a possibility to request the IP address that the GoPro got after connecting to an AP in STA mode and retrieving a DHCP lease.

**Describe the solution you'd like**
A method to request the GoPro's assigned IP address via BLE.
A method to subscribe to IP address changes via BLE.

**Describe alternatives you've considered**
I know that it is already possible to set a static IP for the GoPro with RequestConnectionNew, but this is impractical for our scenario, since we are not in control of the AP where the GoPro should connect to and thus do not know the allowed IP ranges of the network.

**Additional context**
See my post for our use case here: https://github.com/gopro/OpenGoPro/issues/240#issuecomment-1535892397

tcamise-gpsw, 2023-12-06:
As per the linked ticket, I'm hopeful that the COHN feature will unlock you here.
In regard to this specific ticket, the [COHN status](https://gopro.github.io/OpenGoPro/http_2_0#get-cohn-status) does provide the IP address

tcamise-gpsw, 2024-04-12:
This is now possible, when COHN is provisioned, via [Get COHN Status](https://gopro.github.io/OpenGoPro/ble/features/cohn.html#get-cohn-status) as its response Protobuf object contains an [ipaddress field](https://gopro.github.io/OpenGoPro/ble/protocol/protobuf.html#notifycohnstatus)

closed 2024-04-12 by tcamise-gpsw