gopro/OpenGoPro #351: Which (if any) time protocol is used by MAX?

Jemsann, 2023-06-05:
I'm using a MAX in a thesis report and am wondering about time synchronization.

If the datetime is set through BLE or HTTP, will the camera ask for datetime updates after the initial setup or is it a one-time setup? Does the camera use any type of protocol like NTP?  

I have the same question about setting up the datetime through GPS. Will the camera update datetime continuously or just once?

KonradIT, 2023-06-05:
Typically, it's up to the user to sync the camera's time using BLE API or WiFi API. The camera can't call any NTP server because it cannot reach the internet. The GoPro App for phones sends a date time update command every time it connects to the camera. GoPro Labs offers the possibility of using GPS time as onboard clock. MAX not supported. https://gopro.github.io/labs/control/gpssync/

Jemsann, 2023-06-05:
I understand that the camera cant use a NTP server, but does it ask for datetime updates directly to the phone if the user has set up datetime through BLE API or WiFi API? Or is that type of communication only one-way (meaning from the phone to the camera)?

It seems that the MAX is GPS enabled, but do you mean that the datetime in the camera cant be set from the GPS signal?

KonradIT, 2023-06-05:
It doesn't ask for any updates afaik. You can't expect the client to have a real time clock. Rather, the client *may* send the date time, such as the phone app sending it once the camera is connected. This specific GoPro Labs feature is not supported on the MAX, even though the MAX has a GPS antenna.

tcamise-gpsw, 2023-12-06:
Note that MAX is not [supported by Open GoPro](https://gopro.github.io/OpenGoPro/http_2_0#supported-cameras).
The information above does apply to the supported cameras. In any case I'm closing this since there is nothing actionable.