gopro/OpenGoPro #352: Triggering shutter stops the preview stream

follesoe, 2023-06-08:
We are experimenting with the OpenGoPro APIs over USB.

If we issue `/gopro/camera/shutter/start`, while the preview stream is running (`/gopro/camera/stream/start`) it stops. After calling stop on the shutter `/gopro/camera/shutter/stop`, we have to also stop the non-working stream and re-start it every time.

The documentation does not mention that the preview stream will be unavailable if recording/triggering with the shutter. We are using a GoPro Hero 11 Mini, running firmware 2.0.

follesoe, 2023-06-08:
Found this on the Home page describing the live streaming options.. Which is a major bummer, as we would love to use the preview stream while recording/taking photos.

[image]

KonradIT, 2023-06-08:
This is a known issue since a few generations back, at least Hero5 iirc.

follesoe, 2023-06-09:
What about still pictures? Is it possible to capture still pictures without interfering with the live stream? We are testing with a GoPro Hero 11 Mini which doesn't support pictures, so have not been able to test yet.

cheind, 2024-03-15:
@follesoe are you saying that you can sucessfully restart the preview stream after triggering the shutter? I'm testing this on a Hero12 using the WiredGoPro (USB) interface and it seems that the API reports a successful restart of the stream, but the client (ffplay or vlc) is not able to connect again.

aveary, 2024-03-15:
would love a way to stream multiple cameras to a computer and start/shutter without interrupting the stream. This possible yet?

cheind, 2024-03-18:
> @follesoe are you saying that you can sucessfully restart the preview stream after triggering the shutter? I'm testing this on a Hero12 using the WiredGoPro (USB) interface and it seems that the API [...]

 just figured out that works fine. one needs to switch to video mode for the preview to work. hence if you want cap a photo you need to: a) disable preview, b) switch photo, c) shutter, d) wait for ready, e) switch to video, f) enable preview

follesoe, 2024-03-19:
good find @cheind thanks for updating.

tcamise-gpsw, 2024-04-12:
It's good that you have all found a workaround but I will leave this ticket open as an enhancement request to allow encoding during preview stream.

follesoe, 2024-09-05:
I also see that the documentation has been updated to reflect that preview streaming while recording is supported on Hero 12 and upwards.

pavlosharhan2, 2024-09-30:
> just figured out that works fine. one needs to switch to video mode for the preview to work. hence if you want cap a photo you need to: a) disable preview, b) switch photo, c) shutter, d) wait for [...]

Do you mean that for the workaround to work you need to stop the preview stream every time recording a video or taking a photo?

cheind, 2024-09-30:
@pavlosharhan2, yes I needed to stop the preview stream everytime before I could take a photo. At least this was the case for Hero12/firmware 2.0 I was using at the time. Not sure if anything has changed in the meantime?