gopro/OpenGoPro #361: Timeout Error while connect BLE

1265775896, 2023-06-27:
**Component**
What is the bug in?
Python Demo

**Describe the bug**
Timeout Error while connect BLE in every wireless command
Wired command like gopro-webcam works fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow https://gopro.github.io/OpenGoPro/python_sdk/installation.html
2. Execute any wireless command
3. Stuck on "Establishing BLE connection"
4. Gopro displayed "connection successful" and nothing happened.
5. See error

**Logs**
<details>
  <summary>Execute gopro-livestream</summary>

```
   MainThread:19:31:54.495 wireless.py                                95 DEBUG    | Wifi setup. Using [Wireless] driver::[wlp3s0] interface
         data:19:31:54.496 bleak_wrapper.py                          158 INFO     | Scanning for GoPro [A-Z0-9]{4} bluetooth devices...
         data:19:31:54.954 bleak_wrapper.py                          173 INFO     | 	Discovered: E0:41:55:C6:EA:FB: GoPro 8964
         data:19:31:54.961 bleak_wrapper.py                          182 INFO     | Found 1 matching devices.
   MainThread:19:31:54.962 client.py                                 105 INFO     | Establishing the BLE connection
         data:19:31:54.963 bleak_wrapper.py                          252 INFO     | Establishing BLE connection to E0:41:55:C6:EA:FB: GoPro 8964...
   MainThread:19:32:27.402 gopro_wireless.py                         326 ERROR    | Error while opening: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7f15ee6970d0>
command: /usr/bin/bluetoothctl
args: ['/usr/bin/bluetoothctl']
buffer (last 100 chars): b'EL\x01\x1b[0m\x02] Device E0:41:55:C6:EA:FB GoPro 8964\r\n\x1b[0;94m[GoPro 8964]\x1b[0m# \r\x1b[0;94m[bluetooth]\x1b[0m# \x1b[K'
before (last 100 chars): b'EL\x01\x1b[0m\x02] Device E0:41:55:C6:EA:FB GoPro 8964\r\n\x1b[0;94m[GoPro 8964]\x1b[0m# \r\x1b[0;94m[bluetooth]\x1b[0m# \x1b[K'
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 58857
child_fd: 11
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'Accept pairing')
   MainThread:19:32:27.416 client.py                                 129 INFO     | Terminating the BLE connection
         data:19:32:27.419 bleak_wrapper.py                          456 INFO     | Disconnecting...
   MainThread:19:32:27.421 photo.py                                   51 ERROR    | BleakDBusError('org.freedesktop.DBus.Error.UnknownObject', 'Method "Disconnect" with signature "" on interface "org.bluez.Device1" doesn\'t exist\n')
```

</details>

<details>
  <summary>Execute gopro-photo</summary>

```
   MainThread:19:33:49.078 wireless.py                                95 DEBUG    | Wifi setup. Using [Wireless] driver::[wlp3s0] interface
         data:19:33:49.079 bleak_wrapper.py                          158 INFO     | Scanning for GoPro [A-Z0-9]{4} bluetooth devices...
         data:19:33:49.246 bleak_wrapper.py                          173 INFO     | 	Discovered: E0:41:55:C6:EA:FB: GoPro 8964
         data:19:33:49.252 bleak_wrapper.py                          182 INFO     | Found 1 matching devices.
   MainThread:19:33:49.253 client.py                                 105 INFO     | Establishing the BLE connection
         data:19:33:49.254 bleak_wrapper.py                          252 INFO     | Establishing BLE connection to E0:41:55:C6:EA:FB: GoPro 8964...
   MainThread:19:34:21.338 gopro_wireless.py                         326 ERROR    | Error while opening: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7f88dd1980d0>
command: /usr/bin/bluetoothctl
args: ['/usr/bin/bluetoothctl']
buffer (last 100 chars): b':55:C6:EA:FB Paired: yes\r\n\x1b[0;94m[GoPro 8933]\x1b[0m# \r\x1b[KPairing successful\r\n\x1b[0;94m[GoPro 8964]\x1b[0m# '
before (last 100 chars): b':55:C6:EA:FB Paired: yes\r\n\x1b[0;94m[GoPro 8933]\x1b[0m# \r\x1b[KPairing successful\r\n\x1b[0;94m[GoPro 8964]\x1b[0m# '
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 59345
child_fd: 11
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'Accept pairing')
   MainThread:19:34:21.342 client.py                                 129 INFO     | Terminating the BLE connection
         data:19:34:21.343 bleak_wrapper.py                          456 INFO     | Disconnecting...
         data:19:34:23.448 bleak_wrapper.py                          459 INFO     | Device disconnected!
   MainThread:19:34:23.449 photo.py                                   51 ERROR    | TIMEOUT("Timeout exceeded.\n<pexpect.pty_spawn.spawn object at 0x7f88dd1980d0>\ncommand: /usr/bin/bluetoothctl\nargs: ['/usr/bin/bluetoothctl']\nbuffer (last 100 chars): b':55:C6:EA:FB Paired: yes\\r\\n\\x1b[0;94m[GoPro 8964]\\x1b[0m# \\r\\x1b[KPairing successful\\r\\n\\x1b[0;94m[GoPro 8964]\\x1b[0m# '\nbefore (last 100 chars): b':55:C6:EA:FB Paired: yes\\r\\n\\x1b[0;94m[GoPro 8964]\\x1b[0m# \\r\\x1b[KPairing successful\\r\\n\\x1b[0;94m[GoPro 8964]\\x1b[0m# '\nafter: <class 'pexpect.exceptions.TIMEOUT'>\nmatch: None\nmatch_index: None\nexitstatus: None\nflag_eof: False\npid: 59345\nchild_fd: 11\nclosed: False\ntimeout: 30\ndelimiter: <class 'pexpect.exceptions.EOF'>\nlogfile: None\nlogfile_read: None\nlogfile_send: None\nmaxread: 2000\nignorecase: False\nsearchwindowsize: None\ndelaybeforesend: 0.05\ndelayafterclose: 0.1\ndelayafterterminate: 0.1\nsearcher: searcher_re:\n    0: re.compile(b'Accept pairing')")

```

</details>

**Hardware**
- Hero9
- Python3.9
- ArchLinux

tcamise-gpsw, 2023-12-06:
This pexpect interface for Bluez is not very robust. It has currently only been tested on Ubuntu and RPI.
I'll need to replicate this to comment further.

tcamise-gpsw, 2023-12-07:
I've added logging to the pexpect interface. If you can run again from the `main` branch and post the `gopro_demo.log` file that is generated when the failure occurs, I should be able to see what is happening.

closed 2024-01-18 by github-actions[bot]