gopro/OpenGoPro #368: GoPro 11 USB connection to computer disables external microphones

erbensley, 2023-07-18:
**Component**
What is the bug in?
Camera Firmware (probably)

**Describe the bug**
When using a media mod or 3.5mm mic adapter to connect the GoPro Hero 11 to an external audio source, plugging it in to my computer via USB disables the external audio source.  This prevents use of OpenGoPro to control cameras using line-in audio with the mentioned peripherals.

**To Reproduce**
Steps to reproduce the behavior:
1.  Set up GoPro with external audio source via 3.5mm mic adapter or media mod
2.  Capture and notice that the external audio source works
3.  Connect gopro via USB (through the 3.5mm mic adapter or media mod USB connection) to your desktop PC
2.  Capture via OpenGoPro and notice that the captured video has no audio.

**Expected behavior**
I should be able to use the OpenGoPro library to capture on a USB connected GoPro with line-in audio.

**Hardware**
- GoProHero 11 with GoPro labs 2.10.70 firmware (latest as of this writing)
- GoPro Media Mod or GoPro 3.5mm mic adapter (either)
- Desktop (tested on Ubuntu 22.04)

**Additional context**
Add any other context about the problem here.

erbensley, 2023-07-20:
Submitted to https://github.com/gopro/labs/issues/455 as well

erbensley, 2023-07-20:
I did a little more testing and validated the same behavior occurs when camera is connected to a Windows desktop, but does not occur when plugged into a USB power source.  After diving into this a little further I suspect the problem is tied to the underlying implementation of the "Enable wired usb control" API functionality as described in https://gopro.github.io/OpenGoPro/http_2_0

Specifically the following behavior seems to make it very clear the problem is firmware and not client software because of the following sequence
1.  Plug gopro with media mod or 3.5mm adapter into a desktop and connect a 3.5mm audio source as well
2.  Notice that gopro is now nonresponsive, screen just displays "USB connected"
3.  Enable wired USB control via the API
4. Camera can now be controlled manually as well
5.  Perform a capture via the opengopro python library
6.  Notice the lack of audio
7. Perform a capture via manual manipulation of camera controls
8. Notice the lack of audio
9.  unplug from computer (but not from audio source)
10. perform a capture manually
11. notice the audio

tcamise-gpsw, 2023-12-08:
Closing since this is a hardware limitation and thus not solvable via Open GoPro

closed 2023-12-08 by tcamise-gpsw