gopro/OpenGoPro #374: Can not change FOV to others than wide in livestream mode

mahboobkarimian, 2023-07-20:
**Component**
What is the bug in?
- Demos (python SDK)
Look at:
https://github.com/gopro/OpenGoPro/blob/5baddd029d87abbfd6e0a0b578e815316e121e91/demos/python/sdk_wireless_camera_control/open_gopro/demos/gui/livestream.py#L109

**Describe the bug**
In livestream, Setting FOV to linear or superview does not have any effect. It stays on Wide.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `livestream.py` with any described values available at `live_streaming.proto`.
2. You will get always a wide-angle stream!

**Expected behavior**
I expect to change the lens type to something other than wide.

**Screenshots**
No.

**Hardware**
- Camera: Hero9
- Desktop: Linux
- Raspberry Pi 4

**Additional context**
This is a request: Add 4k streaming resolution to the window size option if possible.

armanddp, 2023-09-02:
I thought I was doing something wrong, also bashing my head against this. 

In the LiveStreamStatusUpdate responses that comes back the liveStreamLensSupported is always set to false, and the liveStreamLensSupportedArray set to empty. 

This is using Swift on iOS. Seems the camera is just ignoring the lens settings.

mahboobkarimian, 2023-09-02:
Yes, I saw also the same. The relevant FOV setting is set to false in response to the query of the available/adjustable settings in the camera.

mahboobkarimian, 2023-12-07:
> Hello. I just tested this on Hero 12 by running the livestream demo once with lens set to `EnumLens.LENS_LINEAR` and once with lens set to `EnumLens.LENS_SUPERVIEW` and I see, as expected, an obvious difference in the FOV.
> 
> So I think it is clear that the FOV argument works when starting livestream via the [set_livestream_mode](set_livestream_mode) command.
> 
> Was that the bug described here? Or are you trying to change the FOV when a livestream has already been started. If the latter, this is not currently supported.

As we stated in the previous 2 posts, the cameras we used (hero9 and hero10 at least) don't allow us to change this.

> In the LiveStreamStatusUpdate responses that comes back the liveStreamLensSupported is always set to false, and the liveStreamLensSupportedArray set to empty.

Yes, we set first the mode and then we start the stream. We don't change it once the stream is started.
Maybe it is more related to firmware than your SDK?

tcamise-gpsw, 2023-12-11:
It turns out that the correct behavior was already discovered [above](https://github.com/gopro/OpenGoPro/issues/374#issuecomment-1703658725).

Older cameras simply do not support configurable FOV for livestreaming. So the correct behavior is to dynamically check the supported lens array in Preset Status to know which FOV's are supported.

There are no plans to backport this functionality to older cameras. However, I will leave this open as a documentation bug until the protobuf documentation is updated.

tcamise-gpsw, 2024-04-12:
The documentation [has been udpated](http://localhost:4998/ble/protocol/protobuf.html#requestsetlivestreammode) to note that the 
 > The set of supported lenses is only available from the live_stream_lens_supported_array in [NotifyLiveStreamStatus](http://localhost:4998/ble/protocol/protobuf.html#proto-notifylivestreamstatus))

closed 2024-04-12 by tcamise-gpsw