gopro/OpenGoPro #379: Pairing process fails

simohn, 2023-08-07:
**Component**
What is the bug in?
- Demos (python sdk, etc)
- Specs (BLE , HTTP)

**Describe the bug**
With newer versions of bluetoothctl the pairing process fails, because the syntax of showing the paired devices has changed from `paired-devices` to `devices Paired`.

**To Reproduce**
Try to pair a gopro with a new version of bluetootctl. (I don't know exactly from which version on the syntax has changed. The version with I have tested it is 5.66)

**Expected behavior**
The pairing process should work.

**Hardware**
- Camera: Hero10Black
- Desktop if applicable: Raspberry Pi with Debian

**Solution**
Code change in bleak_wrapper.py

tcamise-gpsw, 2023-08-15:
what version of bluetoothctl are you using? Try sending `version` inside bluetoothctl

tcamise-gpsw, 2023-08-15:
actually don't worry about my previous question. I think I have a fix to handle both cases

tcamise-gpsw, 2023-08-15:
I have a fix for this and also added some more RPI bluetoothctl support. Once it passes checks, I'll merge and release a new version of the SDK.

This whole bluetoothctl thing is getting kind of hacky but it's really just a temporary solution until bleak supports linux pairing

closed 2023-08-15 by tcamise-gpsw

tcamise-gpsw, 2023-08-15:
fix has been published in [0.13.1 release](https://pypi.org/project/open-gopro/0.13.1/)

simohn, 2023-08-25:
Nice thank you :)