gopro/OpenGoPro #388: Directory counter 100.., 101GOPRO, 102... etc. and hard coded path "100GOPRO" let OpenGoPro stop working after 999 files

robertgalo, 2023-08-28:
Within the library (in particular here in http_commands.py and as an example) 

- get_media_info
- delete_media_file (not officially, by myself)
- add_file_hilight
- remove_file_hilight
- get_gpmf_data
- get_screennail__call__
- get_thumbnail
- download_file

there are various places where the current path for functions is linked with a static, hard coded value for the path to the current files by "100GOPRO". According to official GoPro Information, after 999 files a new folder 101GOPRO is created, after next 999 files 102... and so on. In this case, when reaching 999 files, the OpenGoPro library is not working properly and you can't do anything anymore with your code/solution/program where you rely on a working library. This happens as well, when you always delete your files.

There could be 2 solution fixing this problem:

1. There is a flag in the firmware which can be set by an command and which turns this behavior of and sets always 100GOPRO as main directory

2. There is 
a) a function, which reports the current working directory back and injects it into every necessary path of above functions in the list and
b) a function to choose the path by yourself. 
c) Additionally there should be a function to select the director where the files get saved and the possibility to create own directories.

The-Skunk, 2023-08-29:
There is a request related to this, probably in the labs board: https://github.com/gopro/labs/discussions/505

tcamise-gpsw, 2023-09-14:
Working on this soon. I'll change this to force the user to always enter the complete path, including directory.

The-Skunk, 2023-09-15:
> Working on this soon. I'll change this to force the user to always enter the complete path, including directory.

Cool, thanx. 

The only question would be, how to get the current working directory. I did a fast hack and reused the get_media_list function calling it "get_media_folders" giving me back all folders on cam and by an own, outside function I checked which is the current one by number and date - meaning, which one has the latest files written inside.

tcamise-gpsw, 2023-09-18:
> Cool, thanx.
>
> The only question would be, how to get the current working directory. I did a fast hack and reused the get_media_list function calling it "get_media_folders" giving me back all folders [...]

I've added a post-init step in the [MediaList](https://gopro.github.io/OpenGoPro/python_sdk/api.html#open_gopro.models.media_list.MediaList) class to append the directory name to [each filename](https://gopro.github.io/OpenGoPro/python_sdk/api.html#open_gopro.models.media_list.MediaItem.filename). So you should now get the full filename when accessing the media items through, for example, the [files ](https://gopro.github.io/OpenGoPro/python_sdk/api.html#open_gopro.models.media_list.MediaList.files)helper property.

So for example:
```python
import asyncio

from open_gopro import WirelessGoPro

async def main():
    async with WirelessGoPro() as gopro:
        # Get flat list of all media files
        media_files = (await gopro.http_command.get_media_list()).data.files
        # Get the first file
        media_file = media_files[0]
        print(media_file.filename) # Prints: 100GOPRO/GOPR0001.JPG
        # Download
        await gopro.http_command.download_file(camera_file=media_file.filename)

asyncio.run(main())
```

I'll wait to see if anyone has any issues with this but otherwise will merge it and make a release in a few days.

robertgalo, 2023-09-20:
@tcamise-gpsw , maybe this is also important and, I think, there might be an further issue but I can't figure out where it is and if it can be resolved by your fix:

If I open the camera through mtp (Linux, UBUNTU 20.04 LTS) it shows me two folder, 100GOPRO and 101GOPRO. In every of these folder there are several mp4-files (which are valid videos). If I use http://172.2[X].1[Y][Z].51:8080/gopro/media/list it shows me only the files from the 101GOPRO folder. The OpenGoPro function get_media_list() --> HttpParsers.MediaListParser() returns the same results - only the 101GOPRO folder and its files.

Is there something else or do I miss something to add?

tcamise-gpsw, 2023-09-21:
@robertgalo Your last two comments seem worthwhile to investigate but are out of the scope of this issue. Can you make new issues for them?

Thanks.  I'll merge the PR for this fix now.

closed 2023-09-21 by tcamise-gpsw

The-Skunk, 2023-09-24:
> @robertgalo Your last two comments seem worthwhile to investigate but are out of the scope of this issue. Can you make new issues for them?
> 
> Thanks. I'll merge the PR for this fix now.

Here in or somewhere else (I don't think that this is an OpenGoPro issue)?

robertgalo, 2023-09-26:
> worthwhile

@tcamise-gpsw yes, I know, I tried just to mention it not to forget it (as well as for myself ;-)) I'll put them into another thread, thx.