gopro/OpenGoPro #389: Bug Description

Rzbck, 2023-08-31:
**Component**
What is the bug in?
- Tutorials

**Describe the bug**
Traceback (most recent call last):
  File "/Users/Data_C0re_/Desktop/GoGoGo.py", line 3, in <module>
    gopro = WirelessGoPro("5815")
  File "/Users/Data_C0re_/anaconda3/envs/GoPro/lib/python3.10/site-packages/open_gopro/gopro_wireless.py", line 166, in __init__
    wifi_controller=wifi_adapter(wifi_interface, password=sudo_password) if enable_wifi else None,
  File "/Users/Data_C0re_/anaconda3/envs/GoPro/lib/python3.10/site-packages/open_gopro/wifi/adapters/wireless.py", line 86, in __init__
    ensure_us_english()
  File "/Users/Data_C0re_/anaconda3/envs/GoPro/lib/python3.10/site-packages/open_gopro/wifi/adapters/wireless.py", line 41, in ensure_us_english
    language = os.environ["LANG"]
  File "/Users/Data_C0re_/anaconda3/envs/GoPro/lib/python3.10/os.py", line 680, in __getitem__
    raise KeyError(key) from None
KeyError: 'LANG'

**To Reproduce**
Do this code :

from open_gopro import WirelessGoPro
gopro = WirelessGoPro("5815")
gopro.open()
gopro.ble_command.set_shutter(Params.Toggle.ENABLE)
gopro.close()c

**Expected behavior**
Working

**Hardware**
- Camera: Hero11
- Desktop : MAC M1

**Additional context**
I try to connect to my Gopro11 on mac yired and wireless but nothing want to work , i try with CLI TO nothing work to , but i run simple exemple , if you have some idea i take it , thanks !!!

sjbgg, 2023-11-22:
I guess on MAC os.environ has another keyword for language. 
You can simply bypass the function: ensure_us_english() if you have already set your language to us_english and it should be fine.

tcamise-gpsw, 2023-12-07:
I'm somewhat confused here. Firstly I don't think this actually relates to the tutorials as they do not include the open_gopro Python package. I'm assuming that you are just trying to run open_gopro demos.

open_gopro does in fact require US English in order to send and parse Wifi CLI communication. That is why the environment is checked for the LANG variable. My mac environment does have a LANG variable. I though that every environment had a LANG variable.

What is the name of your environment variable that specifies the language? If you can provide that I will add support for it in the Wifi driver. 
Worst case I will at least throw a more meaningful error when the LANG variable can not be found.

closed 2024-01-18 by github-actions[bot]