gopro/OpenGoPro #392: Inject gpmf data remotely (ble)

JuanIrache, 2023-09-08:
Is there a plan to enable (or document) injecting gpmf metadata from a bluetooth device? This was working on the Hero6 but I don't think it was ever documented: https://github.com/gopro/gpmf-parser/blob/main/samples/hero6%2Bble.mp4

Now that the new flagship GoPro doesn't have a GPS module, it would be really useful to pair a phone or GPS tracker and embed the GPS data.

armanddp, 2023-09-08:
Plus one for this.

tcamise-gpsw, 2024-08-30:
@untereiner Well #220 was closed as a duplicate. This ticket remains open to track the feature request. But no, unfortunately there is no progress to update here

untereiner, 2024-09-02:
I am aware of no updates… Wondering if something planed in a possible futur in our dimension.