gopro/OpenGoPro #395: Hero 12 pairing procedure under Windows

demiantres, 2023-09-14:
First time pairing:
1.) Select "Pair device" on the GoPro.
2.) Select my Windows machine from the list of Bluetooth devices on the GoPro.
3.) Now I can pair as with previous cameras (Hero 11, ...).

But now comes the problem: When I power off the camera and want to re-connect to the camera from a Windows machine, then I have to go to "Pair device" on the GoPro again (nothing else is necessary). Then the connection works. With Hero 11 it was possible to directly connect to the camera.

On other systems (Android, ...) connections work without going to the "Pair device" screen on the GoPro.

Is there a way to directly connect to the GoPro on Windows, as we can with Hero 11 and older models?

tcamise-gpsw, 2023-09-21:
Can you describe in more detail how you are pairing? Is this just from the OS options in Windows?

demiantres, 2023-09-23:
Yes, it is only by using the Windows Bluetooth manager.

tcamise-gpsw, 2024-04-12:
duplicate of #476