gopro/OpenGoPro #400: SDK is sending the param name instead of the param value??

tony-gutierrez, 2023-09-21:
**Component**
What is the bug in?
- python sdk

**Describe the bug**
The sdk is resulting in http command like:
`http://172.28.151.51:8080/gopro/camera/setting?setting=162&option=MaxLensMode.DEFAULT`
instead of:
`http://172.28.151.51:8080/gopro/camera/setting?setting=162&option=0`

**To Reproduce**
Steps to reproduce the behavior:
1. Run the video demo with logging level set to DEBUG with cam connected to USB
2. Observe the URLs called

**Expected behavior**
Correct url

**Hardware**
- Camera: Hero11 black
- Desktop if applicable MAC

tony-gutierrez, 2023-09-21:
Looks like this fixes the issue

[image]

tony-gutierrez, 2023-09-21:
@tcamise-gpsw

tcamise-gpsw, 2023-09-21:
Thanks, this really should not have slipped by. Let me add some unit testing for regression and open a PR

tony-gutierrez, 2023-09-21:
It seems like with a bad value, the camera returns a 400, empty `{}`. With a good value but invalid, it returns a 403 with some JSON:
```
{
	"error": 2,
	"setting_id": 173
}
```

I discovered this as the Hero 11 does not seem to support this setting in the video.py demo (at least while on usb):
`await gopro.http_setting.video_performance_mode.set(Params.PerformanceMode.MAX_PERFORMANCE)`
`http://172.28.151.51:8080/gopro/camera/setting?setting=173&option=0`

tcamise-gpsw, 2023-09-21:
Yeah the SDK does not make an effort to know which settings are supported on which cameras / FW.  That's why the demos don't assert on these commands.  They are only set for the relevant firmware versions where this setting mattered.

tcamise-gpsw, 2023-09-21:
I just merged a PR to fix this. I will release 0.14.1 which will include this and other fixes now.

closed 2023-09-21 by tcamise-gpsw