gopro/OpenGoPro #404: App store app submission. Need allowance from GoPro to publish

pavloshargan, 2023-09-21:
I recently developed an iOS app, taking inspiration from a Swift example in this repository. This app facilitates camera control from an iPhone using music playback. However, I've hit a roadblock in publishing it to the Apple Store: they require documentary evidence of my app's compatibility with GoPro cameras.

Following the guidance, I approached Support via the gopro.com website. They directed me to the Contact Us -> Partnership -> Software section where I submitted a request. A month has elapsed since then, but I've received no response. I wonder if the delay could be tied to the launch of the GoPro 12, which I must admit, I adore—especially the live 1080p preview during 5.3k60 recording. It's sparked an idea for another iOS app.

**How the App Works:**
After launching the app and connecting the GoPro via BLE, simply place your phone in your pocket. Voilà! Initiating the Play/Pause on your smartwatch, headphones, or cycling computer triggers the camera to either start/stop recording. Additionally, the camera's status—battery, memory, resolution, and so forth—is displayed in the song title/artist sections of a music player.

The journey of creating this app was exciting. Learning Swift and developing the app took about two months. However, launching it has proven challenging due to the Apple Submission Review and liaison with GoPro. Despite my attempts to assert the use of an open API and adherence to GoPro guidelines, particularly concerning the trademark, the demand remains for documentary proof validating the app's publication and compatibility with GoPro cameras.

By the way, here is a **link to a testflight**:  https://testflight.apple.com/join/BSo7yOC7

@tcamise-gpsw, since this repository aims to assist developers in software creation, any guidance on procuring the required evidence for Apple would be immensely appreciated. I'm contemplating directing the Apple Review team to this thread for validation.

Below is the description of my app for publication, wherein I've mentioned the GoPro trademark:

```
Simply launch the app and place your phone in your pocket. Control the camera using music playback on your smartwatch or cycling computer. When the camera is recording, you'll hear a continuous beep, ensuring you don't have to constantly check its status. It's practical whether the camera is mounted on your helmet or bicycle.
Features:

    Continuous beeping during recording
    Start/stop recording
    View current status and settings
    Add highlights

This app is compatible with GoPro® HERO cameras.
This product and/or service is not affiliated with, endorsed by, or in any way associated with GoPro Inc. or its products and services. GoPro, HERO, and their respective logos are trademarks or registered trademarks of GoPro, Inc.
```

The response I got from Apple review team:
[image]

tcamise-gpsw, 2023-09-22:
Hello. I'm certainly not the correct person to handle this but I will try to start asking around to see if I can find someone to help you.

pavloshargan, 2023-09-22:
Thanks Tim.

Als leaving following here for more clarity:

There are other apps available on the App Store that offer similar functionalities, indicating a precedent for this kind of application:

https://apps.apple.com/us/app/pro-mote/id1548512761
https://apps.apple.com/us/app/bluetooth-remote-for-gopro/id1553411394
https://apps.apple.com/us/app/gowatch-for-gopro/id1569569113

pavloshargan, 2023-10-04:
By the way, here is a video demonstration of how it works: https://www.youtube.com/watch?v=z2LTVEi70LY&t=5s

mattjanus, 2024-02-27:
Isn't this document enough? https://github.com/gopro/OpenGoPro/blob/main/LICENSE

mosquitodu, 2024-08-10:
@pavloshargan did you resolve this issue?

flowstateai-dev, 2024-09-11:
@pavloshargan I have the same question, was the License file enough to resolve the issue?

pavloshargan, 2024-09-12:
> @pavloshargan I have the same question, was the License file enough to resolve the issue?

I gave up on that. Will send submission again tomorrow, will keep you posted

pavloshargan, 2024-12-09:
Hi there!

I've finally made it. The most important move from me when fighting the Apple review was to refer to the MIT license in this repo. I've published the app, also It's open source (IOS/Android) - the repository is in my GitHub account. The IOS app is here: https://apps.apple.com/app/cliptracer/id6482986618

closed 2024-12-09 by pavloshargan

daleclifford, 2025-04-23:
Did it take 18 months to go through this process? Thanks for sharing your insight around the MIT license for the open source library.