gopro/OpenGoPro #408: How to extend settings in the SDK

tony-gutierrez, 2023-09-26:
Is there a way to extend the existing GoPro settings without modifying the SDK? I am new to python, so it might be obvious to others. 

**Describe the solution you'd like**
For example, the SDK is missing Beep control. I would like to extend it to add this setting, without modifying the  SDK source.
Something like:
```
class Beeps(GoProEnum):
            OFF = 0
            ON = 70
            
gopro.http_setting.beeps = HttpSetting[Beeps](communicatior??, SettingId.INTERNAL_87)
assert (await gopro.http_setting.beeps.set(Params.Beeps.ON)).ok
```

**Describe alternatives you've considered**
I am able to accomplish the need by modifying the SDK, but would prefer not to.

tony-gutierrez, 2023-10-26:
@tcamise-gpsw  Any insight?

tcamise-gpsw, 2023-12-06:
No this is not possible nor desired since the Python SDK is only intended to support the public facing OpenGoPro API.

closed 2023-12-06 by tcamise-gpsw

tony-gutierrez, 2023-12-06:
There is a lot of functionality in OpenGoPro that is not reflected in the Python SDK.