gopro/OpenGoPro #409: unhashable type: 'list' when getting media list

tony-gutierrez, 2023-09-27:
**Component**
What is the bug in?
- python sdk

**Describe the bug**
Certain camera states result in an `unhashable type: 'list'` error when attempting something like `media_set_before = set((await gopro.http_command.get_media_list()).data.files)`

It seems to be connected to photo series (taken in burst mode). If I delete that photo series the error goes away. As a side note, the camera failed to delete the series when connected via usb. It would spin briefly without deleting anything. I could delete individual frames, but could not delete all until I unplugged the camera. 

**To Reproduce**
In the browser on the affected camera (/gopro/media/list), I see:
```
{
  "id": "74235298953018024",
  "media": [
    {
      "d": "100GOPRO",
      "fs": [
        {
          "n": "GOPR0296.JPG",
          "cre": "1695119849",
          "mod": "1695119849",
          "s": "9944493"
        },
        {
          "n": "GX010297.MP4",
          "cre": "1695119832",
          "mod": "1695119832",
          "glrv": "3072770",
          "ls": "-1",
          "s": "83715068"
        },
        {
          "n": "GX010298.MP4",
          "cre": "1695121862",
          "mod": "1695121862",
          "glrv": "434000748",
          "ls": "-1",
          "s": "11524142429"
        },
        {
          "n": "GX020298.MP4",
          "cre": "1695122607",
          "mod": "1695122607",
          "glrv": "319053987",
          "ls": "-1",
          "s": "8470300422"
        },
        {
          "n": "GX010299.MP4",
          "cre": "1695125543",
          "mod": "1695125543",
          "glrv": "433969338",
          "ls": "-1",
          "s": "11523191053"
        },
        {
          "n": "G0010300.JPG",
          "g": "1",
          "b": "300",
          "l": "329",
          "cre": "1695745216",
          "mod": "1695745216",
          "s": "76098311",
          "t": "b",
          "m": []
        }
      ]
    }
  ]
[... 42 lines cut ...]
                             }, {                                                                                                                                                               
                                 "filename" : "G0010300.JPG",                                                                                                                                   
                                 "creation_timestamp" : "1695745216",                                                                                                                           
                                 "modified_time" : "1695745216",                                                                                                                                
                                 "group_id" : "1",                                                                                                                                              
                                 "group_size" : "76098311",                                                                                                                                     
                                 "group_first_member_id" : "300",                                                                                                                               
                                 "group_last_member_id" : "329",                                                                                                                                
                                 "group_missing_ids" : [                                                                                                                                        
                                                                                                                                                                                                
                                 ],                                                                                                                                                             
                                 "group_type" : "b",                                                                                                                                            
                             }                                                                                                                                                                  
                         ],                                                                                                                                                                     
                     }                                                                                                                                                                          
                 ],                                                                                                                                                                             
             }
```

**Expected behavior**
No error when retrieving media list

**Hardware**
- Camera: Hero11 Blk
- Desktop OsX

joostplas, 2023-12-06:
I'm running into the same issue. Tony, did you ever find a solution / workaround?

tcamise-gpsw, 2023-12-06:
This is probably due to the `__hash__` of the custom pydantic base model not supporting recursive fields. I'll make it recursive

tcamise-gpsw, 2023-12-06:
The fix seems to be working. I tested taking multiple photo bursts and then getting the media list as described above.

joostplas, 2024-01-20:
Finally had a chance to test the fix. It doesn't yet seem to be in the latest pip version (0.15.1), but when I applied the fix manually in my library it worked like a treat. Much appreciated!