gopro/OpenGoPro #412: Bug Description

hideakidayo, 2023-10-02:
**Component**
What is the bug in?
- Demos (python sdk)
- Specs (HTTP)

**Describe the bug**
test.py
`import subprocess`
`subprocess.run(["gopro-photo", "--wired", "--identifier", "3282", "--output", ". /images/"])`

when running HERO12, you can see the error as shown in the screenshot.
Previously I used HERO11 and with the SDK as of HERO11 this command could be executed without error.

In the SDK as of 0.14.1, I also saw the same error as #409 when I executed the same command in HERO11.

**Expected behavior**
Originally, when this command is executed, the photos taken are saved in the images folder, but an error occurs.

**Screenshots**
[image]

**Hardware**
- Camera: Hero12.
- Windows11
- python3.11.4

jacksongs, 2023-10-26:
I appear to be having similar issues. I can make requests using the python library "requests" to the HTTP endpoints without error, and the endpoints even work in the browser.

Running:
-Python 3.10.11 (via miniconda)
-OSX Sonoma

OUTPUT:

(base) XXXXX % gopro-photo --wired --identifier XXXX
INFO     23:19:05.757                                                                                                     builders.py:853
         <<<<<<<<--------------                                                                                                          
             "id" : "Camera Control Wired Usb",                                                                                          
             "protocol" : "HTTP",                                                                                                        
             "endpoint" : "gopro/camera/control/wired_usb?p=1",                                                                          
             "p" : "Toggle.ENABLE",                                                                                                      
         <<<<<<<<--------------                                                                                                          
                                                                                                                                         
WARNING  23:19:05.774 ConnectionError(ProtocolError('Connection aborted.', BadStatusLine('gpHttpServer: [07353924][INFO gopro_base.py:359
         ] conn 0: /gopro/camera/control/wired_usb?p=1 @civetweb.c:17872\n')))                                                           
WARNING  23:19:07.777 Retrying #0 to send the command...                                                                gopro_base.py:364
WARNING  23:19:07.785 ConnectionError(ProtocolError('Connection aborted.', BadStatusLine('gpHttpServer: [07355935][INFO gopro_base.py:359
         ] conn 0: /gopro/camera/control/wired_usb?p=1 @civetweb.c:17872\n')))                                                           
WARNING  23:19:09.788 Retrying #1 to send the command...                                                                gopro_base.py:364
WARNING  23:19:09.796 ConnectionError(ProtocolError('Connection aborted.', BadStatusLine('gpHttpServer: [07357946][INFO gopro_base.py:359
         ] conn 0: /gopro/camera/control/wired_usb?p=1 @civetweb.c:17872\n')))                                                           
WARNING  23:19:11.800 Retrying #2 to send the command...                                                                gopro_base.py:364
WARNING  23:19:11.807 ConnectionError(ProtocolError('Connection aborted.', BadStatusLine('gpHttpServer: [07359958][INFO gopro_base.py:359
         ] conn 0: /gopro/camera/control/wired_usb?p=1 @civetweb.c:17872\n')))                                                           
WARNING  23:19:13.811 Retrying #3 to send the command...                                                                gopro_base.py:364
WARNING  23:19:13.818 ConnectionError(ProtocolError('Connection aborted.', BadStatusLine('gpHttpServer: [07361969][INFO gopro_base.py:359
         ] conn 0: /gopro/camera/control/wired_usb?p=1 @civetweb.c:17872\n')))                                                           
WARNING  23:19:15.821 Retrying #4 to send the command...                                                                gopro_base.py:364
ERROR    23:19:15.824 ResponseTimeout('Response timeout occurred of 5 seconds')                                               photo.py:53

tony-gutierrez, 2023-10-26:
I am having the same errors. I was about to open an issue but couldn't recreate today, for some reason. 

`ConnectionError(ProtocolError('Connection aborted.', BadStatusLine('gpHttpServer: [07361969][INFO gopro_base.py:359
] conn 0: /gopro/camera/control/wired_usb?p=1 @civetweb.c:17872\n')))`

jacksongs, 2023-10-30:
I'm able to work around this issue by setting the HTTP version to 0.9 in a curl request, ie:

    command = ['curl','--verbose','--http0.9', "http://XXXXX:8080/gopro/camera/shutter/start"]
    result = subprocess.run(command, capture_output=True, text=True)

tony-gutierrez, 2023-12-07:
I haven't seen that error in a while, but the 12s do seem much more temperamental than the 11s. I'm using a lot of them in a commercial setting. Also, if you have the labs firmware loaded, it seems that you have to introduce some delay in between HTTP commands. Otherwise you get 500 errors.

tcamise-gpsw, 2023-12-08:
This appears to be a duplicate of #437 which will have a fix in the next release.

closed 2023-12-08 by tcamise-gpsw