gopro/OpenGoPro #416: Documentation for all possible preset settings

fr33z3, 2023-10-06:
**Is your feature request related to a problem or use case? Please describe.**
I am sending a [RequestGetPresetStatus](https://github.com/gopro/OpenGoPro/blob/main/protobuf/request_get_preset_status.proto) protobuf command and getting the response [NotifyPresetStatus](https://github.com/gopro/OpenGoPro/blob/main/protobuf/preset_status.proto). Everything works alright but the problem I have is that [PresetSetting#id and PresetSetting#value](https://github.com/gopro/OpenGoPro/blob/f36409a5bee01e7405f3a96fb2bc6da1ec34c06b/protobuf/preset_status.proto#L234) is just an int value and it does not have any title_id like in **Preset** model. I've tried to match it if settings reference from documentation but it seems that document is missing some setting ids. Like for example for TimeWarp preset I receive setting with id 111. And there is no description for this setting anywhere. Not even in capabilities file. Is there some exhaustive documentation on preset settings somewhere? Or could we maybe get title_id in **PresetSetting** as we have in **Preset** model?

**Describe the solution you'd like**
Exhaustive documentation for all possible preset settings and meaning of their values.

**Describe alternatives you've considered**
Add title_id for **PresetSetting** the same way as we have for **Preset** model.

**Additional context**
Or does it exist somewhere and I'm just missing it?

tcamise-gpsw, 2023-12-06:
So to my understand this is all currently documented. Is there something I'm missing?

closed 2024-01-18 by github-actions[bot]

pokmis, 2024-04-17:
@fr33z3, are you able to share an outline of how you're forming the command you send here?

_I am sending a [RequestGetPresetStatus](https://github.com/gopro/OpenGoPro/blob/main/protobuf/request_get_preset_status.proto) protobuf command and getting the response [NotifyPresetStatus](https://github.com/gopro/OpenGoPro/blob/main/protobuf/preset_status.proto)._

I'm unable get any response but I'm not sure the command is being formed correctly... 
`let featureIdAndActionId = Data([0xF5, 0x72])
            var request = OpenGopro_RequestGetPresetStatus()
            request.registerPresetStatus = [OpenGopro_EnumRegisterPresetStatus.registerPresetStatusPreset, 
                                            OpenGopro_EnumRegisterPresetStatus.registerPresetStatusPresetGroupArray]
            let requestData  = try request.serializedData()
            let payload = featureIdAndActionId + requestData`