gopro/OpenGoPro #417: INVALID_PARAM When connecting to go pro via BLE during

UneJolieFleur, 2023-10-08:
**Component**
What is the bug in?
- Demos (python sdk, etc)
- Tutorials

**Describe the bug**
The connection with BLE seems to start correctly and it then stops and fail trying to "set date and time" via BLE
[image]

**To Reproduce**
Steps to reproduce the behavior:
Run this code to reproduce:

```
"""Connect to the Wifi AP of a GoPro camera."""

import argparse
import asyncio
import logging
from typing import Optional

from rich.console import Console

from open_gopro import WirelessGoPro
from open_gopro.logger import set_stream_logging_level, setup_logging
from open_gopro.util import add_cli_args_and_parse, ainput

console = Console()

def parse_arguments() -> argparse.Namespace:
    parser = argparse.ArgumentParser(description="Connect to a GoPro camera's Wifi Access Point.")
    return add_cli_args_and_parse(parser)

async def main(args: argparse.Namespace) -> None:
    setup_logging(__name__, args.log)
    gopro: Optional[WirelessGoPro] = None

    async with WirelessGoPro(args.identifier, wifi_interface=args.wifi_interface, sudo_password=args.password) as gopro:
        # Now we only want errors
        set_stream_logging_level(logging.ERROR)

        console.print("\n\n🎆🎇✨ Success!! Wifi AP is connected 📡\n")
        console.print("Send commands as per https://gopro.github.io/OpenGoPro/http")

        await ainput("[blue]Press enter to disconnect Wifi and exit...", console.print)
        console.print("Exiting...")

    if gopro:
        await gopro.close()

# Needed for poetry scripts defined in pyproject.toml
def entrypoint() -> None:
    asyncio.run(main(parse_arguments()))

if __name__ == "__main__":
    entrypoint()
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware**
- Camera: Hero9 black
- Windows10

thomaswelter, 2023-10-09:
I've got the same problem on our Hero 7 Silver

Scopi1, 2023-10-10:
Same error using GoPro 10

Open Gopro - 0.14.1

Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
Kernel: 5.15.0-1038-raspi
Bluez: 5.64-0ubuntu1

Syslog:
`Oct  9 09:11:53 station1 bluetoothd[835]: Starting SDP server
Oct  9 09:11:53 station1 kernel: [  141.454594] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Oct  9 09:11:53 station1 kernel: [  141.454614] Bluetooth: BNEP filters: protocol multicast
Oct  9 09:11:53 station1 kernel: [  141.454629] Bluetooth: BNEP socket layer initialized
Oct  9 09:11:53 station1 bluetoothd[835]: Bluetooth management interface 1.21 initialized
Oct  9 09:11:54 station1 bluetoothd[835]: Failed to set privacy: Rejected (0x0b)
Oct  9 09:11:58 station1 bluetoothd[835]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Oct  9 09:54:48 station1 systemd[980]: Reached target Bluetooth.
Oct  9 09:54:48 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for F7:F5:72:36:24:F3
Oct  9 09:54:50 station1 bluetoothd[835]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Oct  9 09:54:50 station1 bluetoothd[835]: profiles/battery/battery.c:batt_io_ccc_written_cb() Battery Level: notifications not enabled Request attribute has encountered an unlikely error
Oct  9 09:56:11 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for CD:39:5E:46:59:03
Oct  9 09:56:13 station1 bluetoothd[835]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Oct  9 09:56:13 station1 bluetoothd[835]: profiles/battery/battery.c:batt_io_ccc_written_cb() Battery Level: notifications not enabled Request attribute has encountered an unlikely error
Oct  9 09:56:44 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for F7:F5:72:36:24:F3
Oct  9 09:56:46 station1 bluetoothd[835]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Oct  9 09:56:46 station1 bluetoothd[835]: profiles/battery/battery.c:batt_io_ccc_written_cb() Battery Level: notifications not enabled Request attribute has encountered an unlikely error
Oct  9 09:57:16 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for CD:69:42:23:14:7D
Oct  9 09:57:18 station1 bluetoothd[835]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Oct  9 09:57:18 station1 bluetoothd[835]: profiles/battery/battery.c:batt_io_ccc_written_cb() Battery Level: notifications not enabled Request attribute has encountered an unlikely error
Oct  9 09:58:32 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for CD:39:5E:46:59:03
Oct  9 09:58:37 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for CD:69:42:23:14:7D
Oct  9 09:58:54 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for F7:F5:72:36:24:F3
Oct  9 10:03:26 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for C1:53:F7:F0:46:BE
Oct  9 10:05:57 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for F2:1D:3F:3C:0F:1E
Oct  9 10:08:28 station1 bluetoothd[835]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Oct 10 07:52:26 station1 bluetoothd[835]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Oct 10 07:56:31 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for E3:FE:BF:42:DD:D5
Oct 10 07:57:09 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for E3:FE:BF:42:DD:D5
Oct 10 08:14:11 station1 bluetoothd[835]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Oct 10 08:14:42 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for E3:FE:BF:42:DD:D5
Oct 10 08:14:58 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for E3:FE:BF:42:DD:D5
Oct 10 08:17:42 station1 bluetoothd[835]: src/device.c:load_gatt_db() No cache for D2:20:2E:DE:C3:D8
`

Scopi1, 2023-10-18:
Error not thrown when using v 0.13.

I can reproduce the error in 0.13 using ble_command.set_date_time_tz_dst. Error code 2 Invalid param returned.

Error not thrown when using ble_command.set_date_time

savinasz, 2023-10-24:
u can upgrade the open-gopro package, use pip install open-gopro --upgrade .

AnneKnappe4, 2023-10-27:
Hey I updated the open-gopro package but still have the same issue. was anybody able to solve the problem?

mackanjansson, 2023-11-21:
I had the same issue, but then instead followed the git clone step followed by "cd....." and "pip install ."

dtill, 2023-11-29:
same here with GP Hero 9 Black and latest Linux Raspbian.

Installed from sources like discribed here: 
https://gopro.github.io/OpenGoPro/python_sdk/installation.html#from-sources

now it works. :)

> I had the same issue, but then instead followed the git clone step followed by "cd....." and "pip install ."

tcamise-gpsw, 2023-12-06:
I believe this has been fixed but not yet published. In any case I will verify today and publish soon.

tcamise-gpsw, 2023-12-06:
Yes this has been fixed by not asserting on the Set datetime command since it is not supported on older cameras.

closed 2023-12-06 by tcamise-gpsw

Ahmad-197, 2024-05-10:
Hi, 
I am trying capture a photo by using the demo wireless python sdk. However, I am facing the same error as mentioned above. I have tried the solutions mentioned but I am receiving the same error every time.

[image]

What can I do to solve the issue?
P.S. I am using Python 3.10.7  and open-gopro-0.16.1. I have also tried using open-gopro 0.13.1 and 0.11.2.

tcamise-gpsw, 2024-05-10:
> Hi, I am trying capture a photo by using the demo wireless python sdk. However, I am facing the same error as mentioned above. I have tried the solutions mentioned but I am receiving the same error every time.
> 
> Steps to reproduce: Run gopro-photo in the terminal.
> 
> ![image](https://private-user-images.githubusercontent.com/86935871/329561132-43c326e1-762b-4fff-9be1-09aeaebaebde.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUzNjU0OTYsIm5iZiI6MTcxNTM2NTE5NiwicGF0aCI6Ii84NjkzNTg3MS8zMjk1NjExMzItNDNjMzI2ZTEtNzYyYi00ZmZmLTliZTEtMDlhZWFlYmFlYmRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEwVDE4MTk1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdkZGUyM2U5YWRhYTVhYjhkN2NlNmM0NzJjMDRkM2UzNDllOTJmNDVjZjEyMDQyNDNlZWMzNzY0YjNjNTAzN2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.zlRG_ynFe2bVRZT3fZ9HT9Ld8r18bSxnOjg8XzgY4JY)
> 
> What can I do to solve the issue? P.S. I am using Python 3.10.7 and open-gopro-0.16.1. Camera: Hero 11 mini Black. I have also tried using open-gopro 0.13.1 and 0.11.2. I get the following error while using open-gopro 0.11.2.
> 
> ![image](https://private-user-images.githubusercontent.com/86935871/329563569-1234ff3a-4ea3-461b-a753-3200690abced.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUzNjU0OTYsIm5iZiI6MTcxNTM2NTE5NiwicGF0aCI6Ii84NjkzNTg3MS8zMjk1NjM1NjktMTIzNGZmM2EtNGVhMy00NjFiLWE3NTMtMzIwMDY5MGFiY2VkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEwVDE4MTk1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZiMWJkODJmNDdkNzQ1ODBiOTllYjE0MTE2ODdlNzBhOTE5ZDAzYTcyOGZiMGJlNzg3ZTg2NjdlZjhkZDRlODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OstWgUD6MgUgX7pQad_cyvvvaoF6QLicxehuTGUGtxo)

For future reference: this does not actually relate to the ticket here. See #545