gopro/OpenGoPro #421: Cannot get wired connection Hero 11 with Android phone

MS699, 2023-10-23:
**Component**
What is the bug in?
USB/HTTP API

**Describe the bug**
HTTP Get does not work, GoPro can not be pinged

**To Reproduce**
Steps to reproduce the behavior:

1. Connect Gopro to Android phone (usb-c to usb-c cable)
2. HTTP Get: http://172.2x.1yz.51:8080/gopro/camera/control/wired_usb?p=1

**Expected behavior**
Camera can be controlled by HTTP commands

**Hardware**
- Camera: Hero11 firmware 2.12
- Smartphone: Moto g60s Android 12

**Additional context**
I did get the wired usb working on my windows 10 pc (with webcam driver installed).

I did notice that when I connect my phone to the GoPro via USB, the GoPro first says "USB CONNECTED". I then checked my phone's connected devices > USB and saw that the USB was managed by connected device. I switched that so my device manages the USB connection. Then the GoPro shows its controls again on its screens. So this made me think that there my be another setting I should switch on my phone, download something (like the windows driver), or need to do something code wise.

The quik app has no problems using the usb connection.

AlexSkillman10, 2023-10-25:
I am trying to do this exact same thing, but am having an equal lack of success. If anyone could provide any tips or information it would be greatly appreciated.

vateseif, 2024-04-03:
+1 on macos

pavlosharhan2, 2024-07-26:
Most likely It's a problem on a phone side.

For me USB HTTP Api works On Samsung S22, Pixel 5, but doesn't work On Xperia10. Currently investigating,  I think I have to toggle some setting like Ethernet tethering/Usb tethering/enable HAL. Have tried almost everything, but no luck yet

mmarich, 2024-09-10:
Try placing your phone into airplane mode and disabling WiFi. In my experience so far, that will cause the device to start the USB NCM network interface. I have only tested on a Pixel 6 and 7.

pavlosharhan2, 2024-09-10:
@mmarich this doesn't make much sense, right? why wouldn't i be able to use wifi when USB NCM is active...

Means you can't have internet and NCM connection at a time unless you have cellular data

pavlosharhan2, 2024-09-20:
@tcamise-gpsw @MS699 From my observations, the phone has to have USB3.0 port, otherwise the GoPro won't be mounted as ethernet interface no matter which usb configurations are set. Having the same problem on Sony Xperia 10V, but works fine on Pixel5/Samsung S22