gopro/OpenGoPro #422: Bug Description

savinasz, 2023-10-24:
**Component**
What is the bug in?
- Demos (python sdk, etc)

**Describe the bug**
i want to record video from API, but i get this error, what must i do?

**Screenshots**
(gopro) C:\Users\savin\inastek\gopro-cam\gopro\Scripts\src\OpenGoPro\demos\python\sdk_wireless_camera_control\docs>gopro-video 2
INFO     08:55:47.817 Scanning for GoPro [A-Z0-9]{4} bluetooth devices...                             bleak_wrapper.py:119
INFO     08:55:50.483    Discovered: F3:80:95:5C:15:D7: GoPro 2074                                    bleak_wrapper.py:132
INFO     08:55:50.488 Found 1 matching devices.                                                       bleak_wrapper.py:145
INFO     08:55:50.491 Establishing the BLE connection                                                        client.py:106
INFO     08:55:50.492 Establishing BLE connection to F3:80:95:5C:15:D7: GoPro 2074...                 bleak_wrapper.py:212
INFO     08:55:55.398 Discovering characteristics...                                                  bleak_wrapper.py:353
ERROR    08:55:56.118 Error while opening: Could not read Descriptor value for 0036: Protocol Error  gopro_wireless.py:340
         0x05: Insufficient Authentication
INFO     08:55:56.321 Terminating the BLE connection                                                         client.py:130
INFO     08:55:56.339 Disconnecting...                                                                bleak_wrapper.py:408
INFO     08:55:59.547 Device disconnected!                                                            bleak_wrapper.py:411
ERROR    08:55:59.549 BleakError('Could not read Descriptor value for 0036: Protocol Error 0x05: Insufficient  video.py:55
         Authentication')
Exiting...

**Hardware**
- Camera: Hero9
- Desktop [e.g. Windows 10/11]

savinasz, 2023-10-24:
[image]
 
The previous error has been resolved. But I experienced an error as in the picture.

savinasz, 2023-10-25:
[image]

Now error when i try to connect to the Wi-Fi.

savinasz, 2023-10-25:
[image]
finally i can take a picture with gopro-photo in hero 9 💯

AnneKnappe4, 2023-10-27:
Hello Savina, I have the same issue :( How could you enable the WIFI connection? It is very simple with the gopro 11. The connection is established even if I have turned off the camera. But I have many problems with the GoPro 10. I would appreciate your help.

savinasz, 2023-10-30:
@AnneKnappe4 you can reset connection in gopro. or you can downgrade open-gopro version, i use version 0.11.2 (<0.12.0)

awonglk, 2023-10-31:
I have the same error on Gopro 11 Black running on LABS firmware
Resetting connections didn't help
[image]

AnneKnappe4, 2023-11-01:
Hello Savina, thanks for your respond. Unfortunally nothing works :(

savinasz, 2023-11-06:
@awonglk  you can downgrade open0gopro version. use pip install open-gopro==0.13.1

yutachen1983, 2023-11-14:
I think that is a bug. I trace the code and find the issue. The Invalid parameters is using ble command to set the Timezone and DST, and just remark the function "self.ble_command.set_date_time_tz_dst" in the gopro_wireless.py line 318 to 322 at open-gopro 0.14.1, and then you can connect your gopro.

tcamise-gpsw, 2023-12-06:
Duplicate of #417

closed 2023-12-06 by tcamise-gpsw