gopro/OpenGoPro #426: Incorrect Camera Status HTTP Specs

HydeNaut, 2023-10-27:
**Component**
What is the bug in?
- Specs (HTTP)

**Describe the bug**
The status spec table doesn't  list  a Status ID '61' but it comes up when I check camera status

**To Reproduce**
Steps to reproduce the behavior:
1. Call 'GET(http://172.XX.1XX.X1:8080/gopro/camera/state)'
2. Read read the data returned. Ignoring the settings you will get a something like:
`{
  "status": {
    "1": 0,
    "2": 0,
    "3": 0,
    "4": 255,
    "6": 0,
    "8": 0,
    "9": 0,
    "10": 0,
    "11": 0,
    "13": 0,
    "14": 0,
    "17": 1,
    "19": 0,
    "20": 0,
    "22": 0,
    "23": 0,
    "24": 0,
    "27": 0,
    "29": "",
    "30": "MyPro",
    "31": 0,
    "32": 1,
    "33": 0,
    "34": 7336,
    "35": 0,
    "36": 1165,
    "37": 5,
    "38": 1165,
    "39": 5,
    "41": 0,
    "42": 0,
    "45": 0,
    "49": 0,
    "54": 124602941,
    "55": 1,
    "56": 4,
    "58": 0,
    "59": 0,
    "60": 500,
    "61": 2,
    "64": 0,
    "65": 0,
    "66": 100,
    "67": 100,
    "68": 0,
    "69": 0,
    "70": 0,
    "74": 0,
    "75": 0,
    "76": 0,
    "77": 0,
    "78": 0,
    "81": 1,
    "82": 1,
    "83": 0,
    "85": 0,
    "86": 0,
    "88": 0,
    "89": 16,
    "90": 0,
    "91": 0,
    "93": 655360,
    "94": 786432,
    "95": 851968,
    "96": 1001,
    "97": 786432,
    "98": 0,
    "101": 0,
    "102": 0,
    "103": 0,
    "105": 1,
    "106": 0,
    "107": 1639050,
    "108": 0,
    "109": 0,
    "110": 0,
    "111": 0,
    "112": 0,
    "113": 0,
    "114": 0,
    "115": 0,
    "116": 1,
    "117": 128144375
  }
}`
3. Scroll down to '....'
4. See error

**Expected behavior**
That the actual Status IDs match those documented

**Screenshots**
[image]

[image]

[image]

[image]

**Hardware**
- Camera: Hero12 Black
- Desktop: Windows 10

**Additional context**
I am using a Labs Firmware to control power from USB.

tony-gutierrez, 2024-05-09:
As of a few days ago they were pretty well documented here:
https://gopro.github.io/OpenGoPro/http

But now a lot of settings seem to have been removed, and no statuses are documented.

tcamise-gpsw, 2024-05-09:
Whoops. We're in the middle of a major rearchitecture here so there are some growing pains. I'll fix this today

tcamise-gpsw, 2024-05-09:
Actually the settings look correct to me. But I will make sure the statuses get added to the HTTP spec. 

FYI they also exist in the [BLE spec](https://gopro.github.io/OpenGoPro/ble/features/statuses.html) and are currently there.

tcamise-gpsw, 2024-05-09:
This has been fixed and the HTTP statuses are back.

closed 2024-05-09 by tcamise-gpsw