gopro/OpenGoPro #429: question: Is this repository being maintained or open to external contributors?

alestiago, 2023-11-08:
I'm opening this issue to track how maintained is this repository.

The last main commit was in Sep 21 (1 month, 18 days ago). Two weeks ago I created a Pull Request for a documentation change #425 and it has no activity from the maintainers. There are also other community Pull Request, like #413 that has also no activity from the maintainers for more than a month.

I'm wondering if maintaining this repository has stopped. If so, should this be forked?

KonradIT, 2023-11-08:
If there are no HTTP/BLE API changes with newer firmware revisions or new camera generation releases, likely there are no changes uploaded

alestiago, 2023-11-08:
Thanks @KonradIT for the reply. 

I'm not sure if I'm understanding correctly, so in essence this repository will only get updates from the GoPro team when newer firmware gets updated. It will **NOT** get updates when any community contributor submits a change, like for example fixing a typo in the documentation or improving the Python SDK?

tcamise-gpsw, 2023-12-06:
Yes that is mostly correct. The only thing that is guaranteed to be updated is the documentation, i.e. BLE and HTTP specs.

Everything else is done as best effort.

closed 2023-12-06 by tcamise-gpsw