gopro/OpenGoPro #431: MISC folder access over HTTP command

NickPronto, 2023-11-10:
Feature:
Accessing VIDEOS/DCIM/ (MP4, LRV) and images (THM) are possible through HTTP requests through USB and WIFI connection. 

Acceing VIDEOS/MISC does not seem possible through the same methods.

Solution:
I'd like the access to MISC folder's contents on the SD card when in camera, similar to the DCIM folder access.

Existing access to DCIM folder:
```python
    def download_file(self, filename, out_dir):
        #global filename
        global currenttime
        justfilename = str(filename.rsplit( ".", 1 )[ 0 ])
        src1 = str(self.gopro_url)+"videos/DCIM/100GOPRO/" + justfilename + ".MP4" #mp4
        src2 = str(self.gopro_url)+"videos/DCIM/100GOPRO/" + justfilename + ".THM" #thm
        src3 = str(self.gopro_url)+"videos/DCIM/100GOPRO/" + justfilename.replace("X","L") + ".LRV" #lrv

        try:
            urlretrieve(src1, out_dir+".mp4")
            print("%s downloaded" % out_dir+".mp4")
            urlretrieve(src2, out_dir+".thm")
            print("%s downloaded" % out_dir+".thm")
            urlretrieve(src3, out_dir+".lrv")
            print("%s downloaded"  % out_dir+".lrv")
            return True
        except Exception as e:
            return False
``` 

Requested functionality:

```python
    def download_settings_file(self):
        filename = str(self.gopro_url)+"videos/MISC/GoPro-owner.txt"
        try:
            urlretrieve(filename, "newsettings.txt")
            print("Gopro settings downloaded")
            return true
        except Exception as e:
            return False
``` 

**Additional context**
If this already exists, please let us know. We've been unable to get any function to access the MISC folder's contents over HTTP requests.

tcamise-gpsw, 2023-12-07:
Hello. This is currently purposefully disabled. What is the use case here?

NickPronto, 2023-12-07:
The use case is to access the owner.txt file for download. 

We have figured a way around it, by passing fourCC strings into the video metadata and reading that into a file, but access to the owner.txt file or a similar file in the video folder would make things even better.

closed 2023-12-11 by tcamise-gpsw

Fougl, 2024-03-29:
Hi @tcamise-gpsw , I would love this feature, the use case would be to use the gopro labs motion detecion with the mask file. I would be able to pass remotely the mask file without deinstalling the camera and taking out the sd card. would make the testing of optimal mask immensly more efficient

Fougl, 2024-05-03:
Hi @tcamise-gpsw , any chance we will see this in the future? I am constantly installing cameras, and need to test the masks appropriate for the triggers

pavlosharhan2, 2024-09-25:
+1 for misc folder access for changing the mask dynamically