gopro/OpenGoPro #433: Unable to open HTTP webcam feed

AlsterDev, 2023-11-23:
Hi, I am trying to view a HTTP webcam feed from a Hero 11 Black (version 02.20) on a Win 11 PC using the Python SDK multi_webcam single_webcam.py script, however I get the 'Starting player' and 'Player started' messages, but no player opens. Using the udp://0.0.0.0:8554 address in VLC it appears to connect (well, doesn't throw an unable to connect error) however never loads.

To reproduce the issue, I run
`python single_webcam.py 338`
which gives the output

> DEBUG:root:Using port 8554
> DEBUG:root:Sending camera/control/wired_usb?p=0: {}
> DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 172.23.138.51:8080
> DEBUG:urllib3.connectionpool:http://172.23.138.51:8080 "GET /gopro/camera/control/wired_usb?p=0 HTTP/1.1" 200 3
> DEBUG:root:HTTP return code 200
> DEBUG:root:{}
> INFO:root:Starting webcam
> DEBUG:root:Sending webcam/start: {'port': 8554}
> DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 172.23.138.51:8080
> DEBUG:urllib3.connectionpool:http://172.23.138.51:8080 "GET /gopro/webcam/start?port=8554 HTTP/1.1" 200 31
> DEBUG:root:HTTP return code 200
> DEBUG:root:{
>     "status": 2,
>     "error": 0
> }
> INFO:root:Webcam started successfully
> INFO:root:Starting player @ udp://0.0.0.0:8554
> INFO:root:Player started.
> Press enter to stop

but without a player. The GoPro itself does go into webcam mode, indicating that it thinks its sending a feed, but the PC isn't able to display one.

Any ideas where i might be going wrong?

AlsterDev, 2023-12-07:
Update: Seems like it was just a simple case of Windows' firewall blocking the port. For some reason it wouldn't work with the firewall disabled but creating an inbound rule to allow the port fixed the issue.

tcamise-gpsw, 2023-12-07:
Glad you figured it out. This seems out of the scope of the Python SDK so I'll close this without further action.

closed 2023-12-07 by tcamise-gpsw